realsenseai/librealsense #2381: Please include instructions on using PyPI / pip for python users

BrendanDrewIntel, 2018-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | All | 
| Firmware Version                   | All | 
| Operating System & Version |   Linux (probably all)  | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 |  Ubuntu 16.04 on i7  |
| SDK Version                            |  Current master                          | 
| Language                            |  python  | 
| Segment			|  All | 

### Issue Description
Please update documentation for python users to highlight the fact that pyrealsense2 (https://pypi.org/project/pyrealsense2/) is available via PyPI / installable via pip -- this is an awesome feature and more folks should know about it!

MartyG-RealSense, 2018-09-12:
Hi Brendan!  Yes, the pip installation process is referenced in the installation instructions of the SDK's Python wrapper page but not on the page that you linked to.  Thanks for highlighting this!

dorodnic, 2018-09-12:
@BrendanDrewIntel any additions to the documentation would be greatly appreciated.

closed 2018-10-14 by dorodnic