realsenseai/librealsense #2382: incrorrect dimensiones in space

mat6778, 2018-09-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  WIN 10 64-bit                                        | 
| Platform                                 |PC                                         |
| Language                            |  Python 2.7.15                      | 

### Issue Description
**incorrect dinemsiones in space**
[image]
[image]
Hello Git-Hub-Community,

I made a short programm that should get me the distance between two points in space. To check the accuracy I made a "chessboard" where each edge is exactly 50mm long. In the programm I can choose a place near the edges and the "GoodFeaturesToTrack"-function provides the exact pixels. After that I meassured the distance to the points and used "rs.rs2_deproject_pixel_to_point()" to find the koordinates in the 3D-space. The distance from the D435 to the points is quite accurate but the distance between the pints is always about 40% further than it really is. So i get a distance of 70mm when meassuring the 50mm edges.
Is there something wrong with the camera options?

At this example: https://github.com/IntelRealSense/librealsense/tree/master/examples/measure
there is a function to fill zero pixels and to choose the "High Density"-configuration. How is this possible in Python?

Thank you so much for your help! Be patient! I did not know the term "Python" two weeks ago.

Best regards
Matthias

I attached my code and the pictures.
[image]
[image]
[image]

MartyG-RealSense, 2018-09-12:
I have seen a few cases where there is a similar error ratio when measuring the X & Y coordinates (for example, 7 cm instead of 5 cm, and 80 mm instead of 50 mm).  In the cases where this occurs, greater accuracy is usually achieved by using the camera intrinsics for the measuring process.

There was a Python user over on the Intel Support forum for RealSense who had this problem this week, and they posted their solution.

https://communities.intel.com/message/566043#566043

mat6778, 2018-09-12:
Awsome! This worked at once! I now get the exact dimension -+1mm :)

[image]

MartyG-RealSense, 2018-09-12:
You're very welcome.  Great news that the solution worked for you!  :)

ev-mp, 2018-09-12:
Imo some clarification is required as while the applied method is correct, the provided explanation seem to suggest some empiric evidence, which is misleading:

The de-projection function is deterministic and must use the intrinsic of the frame upon each it is invoked.

In the above scenario using the `depth_intrinsic` for de-projection was incorrect as the pixels were calculated based on the  `aligned_depth` frame, thus "pairing" data from two unrelated frames could not produce the anticipated results. To be consistent the `aligned_intrinsic` (which is identical to  `color_image`) should have been used instead. 

In order to avoid further confusion I would recommend to replace `color_intrinsic`  with 'aligned_intrinsic' in the code to clarify the intend.

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
Hi @mat6778,

You can use the example code like below to set the preset to High Density.

    preset_range = depth_sensor.get_option_range(rs.option.visual_preset)
    for i in range(int(preset_range.max)):
        visulpreset = depth_sensor.get_option_value_description(rs.option.visual_preset,i)
        if visulpreset == "High Density":
            depth_sensor.set_option(rs.option.visual_preset, i)

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @mat6778,

Still need any support for this topic?

closed 2018-11-19 by RealSense-Customer-Engineering