realsenseai/librealsense #2383: 3d printer files for Multipl RealSense camera holders?

iotodev, 2018-09-12:
Looks like I have to experiment with multiple D435 cameras for instantaneous capture.
 
On the recent webinar "Hands-on with Multiple Intel RealSense D415 & D435 camera systems" I noticed what appeared to be 3d printed holders for multiple RealSense camera.
 
Any chance of sharing those files ?  .stl or similar would be grand.  Likely I'll need outward facing, but anything might help
 
Might save me some grief! 
 
Thanks in advance!

dorodnic, 2018-09-13:
@agrunnet any chance you could share?

agrunnet, 2018-09-13:
The files will be available here within one day or so. 
 https://www.intel.com/content/www/us/en/support/articles/000030540.html

closed 2018-09-13 by iotodev