realsenseai/librealsense #2384: save frameset to files for post-processing

ywjia, 2018-09-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10..3.0 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.16.0                          | 
| Language                            |  C++                         | 
| Segment			|  VR                          | 

### Issue Description

I have a question about saving data to file.

Is it possible to save frameset as whole object, as below:

rs2::frameset frameset_w = pipe.wait_for_frames();
save_frameset("frameset_0000.bin", frameset_w);
...

then, in a separate post-processing step (even without camera connected), we can simply do the following to retrieve all the data

rs2::frameset frameset_r = read_frameset("frameset_0000.bin");
rs2::video_frame color = frameset_r.get_color_frame();
rs2::depth_frame depth = frameset_r.get_depth_frame();
...

Furthermore, is it possible to get all other data, such as intrinsic and depth_scale, from frameset_r?

Note that this is not recording the data continuously, since here the user can choose which framesets to save.

Let me elaborate a bit more.

In some use case, I want to take a number of shots of a scene and save all the relevant data (color, depth, intrinsic, depth_scale, et al) into files, and then later I select some shots for post-processing. In the capturing step, I want to capture individual shots, rather than recording the whole scene continuously. In the post-processing step, I still want to be able to access all the rs2 libraries,

What is the best way of doing this?

RealSense-Customer-Engineering, 2018-09-19:
[Realsense Customer Engineering Team Comment]
Hi @ywjia,

You can refer to #1942 to see if it can fulfill your request.

ywjia, 2018-09-19:
Thanks. I read through #1942. It is about keeping multiple frames (color, depth) in memory for later processing in the same program. What I am thinking here is to save all the data related to a frameset into a file, for later processing offline. So, it is a bit different. 

I understand that I can record all the data to a bag file, which tends to be very large. It would be nice to have a method of just dumping one frameset with all the relevant data.

ywjia, 2018-09-21:
thanks. It will be helpful to have this feature.

closed 2018-09-21 by ywjia