realsenseai/librealsense #2386: RGB image turns to all black after upgrading sdk and firmware

chunhai, 2018-09-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.10.03 | 
| Operating System & Version |   Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  4.14.69                                         | 
| Platform                                 | Odroid  |
| SDK Version                            |  v2.15.0/v2.16.0                           | 
| Language                            |  C++                           | 
| Segment			|                          | 

### Issue Description
I tried to upgraded sdk to latest version, so I recompiled with v2.16.0 (https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) and also upgraded the firmware to 5.10.03. However, the RGB image turns to black while depth image is normal. I then tried to recompile with v2.15.0, nothing changes. It's the same issue for both D415 and D435 I have.

Does there anyone know how to fix this? Thanks.

RealSense-Customer-Engineering, 2018-09-12:
[Realsense Customer Engineering Team Comment]
Hi @chunhai,

It looks this issue only happens on your Odroid board? Is it Odroid XU4?
How about the previous librealsense version like 2.14.1?

chunhai, 2018-09-13:
Yes, it is Odroid XU4. The version I used before works fine, but I don't
remember the exact version number, it was installed more than half year
ago. I will try 2.14.1 or even earlier version. Thanks for the reply.

RealSense Customer Engineering <notifications@github.com> 于2018年9月12日周三

ev-mp, 2018-09-13:
A duplicate of #2306.
This ticket will be closed and followed on the original report thread.

closed 2018-09-13 by ev-mp