realsenseai/librealsense #2389: Realsense interface with Google Cartographer demo

ShettyHarapanahalli, 2018-09-13:
Google Cartographer or other SLAM interfaces on ROS

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.10.03.00| 
| Operating System & Version  |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | librealsense2                       | 
| Language                            |  {C/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have recently started using the real sense, I was able to update the firmware on the real sense and launch the viewer application on Ubuntu. 

I want to use real sense for the slam, do we have any example that directly interfaces with the Google Cartographer for other slam applications. I did see the "realsense_samples_ros" in git hub this is archived.

Please share an example which can support the cartographer or other SLAM's.

MartyG-RealSense, 2018-09-13:
In Intel's article about creating a RealSense-equipped mobile robot, it states that although the system uses the open-source 'slam_gmapping' package by default, other SLAM solutions such as Cartographer and rgbd-slam could be used with it instead.

https://software.intel.com/en-us/articles/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr

ShettyHarapanahalli, 2018-09-13:
Well Google cartographer guys said they do not support Intel Real sense for few reasons.
You guys attend the next open house for cartographer to understand more.    
https://github.com/googlecartographer/cartographer [SLAM from Google]

But any ways I got few leads from this thread :  
https://github.com/introlab/rtabmap/issues/262

ShettyHarapanahalli, 2018-09-17:
http://wiki.ros.org/turtlebot3 also integrates the Realsense this is a good information for for the rover.

closed 2018-09-17 by ShettyHarapanahalli

stefangordon, 2018-11-20:
Intel is still advertising that you can use this camera with Google Cartographer, but I haven't seen any evidence yet.  Was a demo ever put together to show how to do this?

ShettyHarapanahalli, 2018-12-04:
We can do this by converting the point cloud data to laser scan with few ROS package like pointcloud_to_laserscan, but the whole point of 3D data is lost with that.

The other interesting point is most of the navigation algorithms that we have are based on 2d.

If we had stable 3d navigation algorithms then the DARPA Subterranean (SubT) Challenge would have not been there[https://www.darpa.mil/program/darpa-subterranean-challenge]. kidding