realsenseai/librealsense #2391: MultiCamera Sync at 25Hz

joemenke, 2018-09-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (5.10.03) | 
| Operating System & Version |  MacOS  |                                 | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.0 }                          | 
| Language                            |  C++                           | 
| Segment			|  VR/AR                | 

Does hardware synchronization work at 25Hz framerate? The reason I ask is that according to the documentation "depth" synchronization is supported in the current firmware, but at 25Hz I can get IR images, but not actual depth images.

MartyG-RealSense, 2018-09-14:
I have done some previous testing of 25 FPS mode.  The number of IR resolutions supported is just two in Y16 format (which is an unrectified channel), and none for the Z16 depth.  This is because 25 FPS is a camera calibration mode and not intended for everyday use in applications.  

A wish has been voiced by some users for support for a 24 FPS mode to help the camera be usable for movie projects, but there are no confirmed plans for that feature to be worked on yet.

You can view the results of my 25 FPS tests at the link below.

https://communities.intel.com/message/557377#557377

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Yes. 25 fps is only for calibration mode. There is no 25 fps depth output. So the multi camera sync does not work for 25 fps.

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hi joemenke,

Do yo have more questions?

closed 2018-11-16 by RealSense-Customer-Engineering