realsenseai/librealsense #2392: uvcvideo Driver Patches for Ubuntu kernel: 4.15.0-34-generic

houlei516, 2018-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

houlei516, 2018-09-14:
Try means and google a lot. I met with the following issue and broke the apt-get. 
[image]

RealSense-Customer-Engineering, 2018-09-17:
[Realsense Customer Engineering Team Comment]
Hi @houlei516,

From your screenshot, it looks you tried to install the prebuilt package 'ros-*-librealsense' which is not for D400 series camera. Probably, this causes the improper installation.

houlei516, 2018-09-17:
I use the sr300. Unplug when build the librealsense. 

I can build from source but I still need to sudo apt get install the dev and etc. It seems building from source is the same to patch the new kernel.

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @houlei516,

"apt-get utils and dkms" should be enough to work for pre-built librealsense apps with SR300 on kernel 4.13 or 4.15. If any other question, let us know. Otherwise, I will close this issue.

closed 2018-09-26 by houlei516