realsenseai/librealsense #2395: Failed installation with linux kernel 4.8.15

aPonza, 2018-09-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 16.04.5 LTS  | 
| Kernel Version (Linux Only)    |  4.8.15-rt10                                         | 
| Platform                                 | PC |
| SDK Version                            |                           | 
| Language                            |                         | 
| Segment			|  Robot                  | 

### Issue Description

I was following the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) and at the step `./scripts/patch-realsense-ubuntu-lts.sh` I end up having this output:

<pre>
ap@ap-NUC7i7DNKE:~/git/librealsense$ ./scripts/patch-realsense-ubuntu-lts.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.4).
The following NEW packages will be installed:
  linux-headers-4.4.0-135 linux-headers-4.4.0-135-generic linux-headers-generic
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 10,9 MB of archives.
After this operation, 78,5 MB of additional disk space will be used.
Get:1 http://it.archive.ubuntu.com/ubuntu xenial-updates/main amd64 linux-headers-4.4.0-135 all 4.4.0-135.161 [10,0 MB]
Get:2 http://it.archive.ubuntu.com/ubuntu xenial-updates/main amd64 linux-headers-4.4.0-135-generic amd64 4.4.0-135.161 [822 kB]                                                                           
Get:3 http://it.archive.ubuntu.com/ubuntu xenial-updates/main amd64 linux-headers-generic amd64 4.4.0.135.141 [2.364 B]                                                                                    
Fetched 10,9 MB in 7s (1.370 kB/s)                                                                                                                                                                         
Selecting previously unselected package linux-headers-4.4.0-135.
(Reading database ... 416098 files and directories currently installed.)
Preparing to unpack .../linux-headers-4.4.0-135_4.4.0-135.161_all.deb ...
Unpacking linux-headers-4.4.0-135 (4.4.0-135.161) ...
Selecting previously unselected package linux-headers-4.4.0-135-generic.
Preparing to unpack .../linux-headers-4.4.0-135-generic_4.4.0-135.161_amd64.deb ...
Unpacking linux-headers-4.4.0-135-generic (4.4.0-135.161) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../linux-headers-generic_4.4.0.135.141_amd64.deb ...
Unpacking linux-headers-generic (4.4.0.135.141) ...
Setting up linux-headers-4.4.0-135 (4.4.0-135.161) ...
Setting up linux-headers-4.4.0-135-generic (4.4.0-135.161) ...
Examining /etc/kernel/header_postinst.d.
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.4.0-135-generic /boot/vmlinuz-4.4.0-135-generic
Setting up linux-headers-generic (4.4.0.135.141) ...
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Cloning into './ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1'...
remote: Counting objects: 60174, done.
remote: Compressing objects: 100% (56328/56328), done.
remote: Total 60174 (delta 5507), reused 19219 (delta 2897)
Receiving objects: 100% (60174/60174), 163.98 MiB | 1.49 MiB/s, done.
Resolving deltas: 100% (5507/5507), done.
Checking connectivity... done.
Note: checking out '2e1f960b96b3d80fc2b91809bf0cb02c97abe8f4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Checking out files: 100% (57407/57407), done.
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1243 (offset 14 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 627 (offset 3 lines).
Applying realsense-metadata patch
./scripts/patch-realsense-ubuntu-lts.sh: line 90: ../scripts/realsense-metadata-ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1.patch: No such file or directory
</pre>

The main problem could be I have a realtime patched kernel and limited availability of those as well, as I need to be compatible with librealsense and libfranka, a third party library for a robotic arm. This means the alternative kernels I could use are 4.16.18-rt12 or 4.13.13-rt5 with increasingly worsening performance results.

Is it possible to continue the installation process? I can't find mentions of 16.04.5 not being supported. I couldn't also find similar issues dealing with a beforehand patched kernel.

ev-mp, 2018-09-16:
Hello @aPonza , we will look into this. 
Kernel 4.8.15-rt10 seem to be a mainline version and not Ubuntu LTS kernel. Is there a repo with a Debian 
 image, or you have to build it from source and then apply PREEMPT-RT patch?

aPonza, 2018-09-17:
I built it from source and then applied the patch following [these](https://frankaemika.github.io/docs/installation.html#setting-up-the-real-time-kernel) instructions.

aPonza, 2018-09-27:
Thank you for the help @ev-mp! I tried again after pulling the changes, and again got blocked at the script. I'm attaching the full [output.txt](https://github.com/IntelRealSense/librealsense/files/2424652/output.txt). It's much better than not starting at all like last time, but still fails.

I was hoping the realsense patch wouldn't touch already patched modules but it seems it is. Hopefully it will keep the preemption. I used default choices all the way down as is likely what was needed, but correct me if this wasn't the case.

Some (Boost?) **warnings** seem to lead to the error, which comes when installing the built modules.

The relevant dmesg returns:
```
[29899.077497] media: Linux media interface: v0.10
[29899.080004] videodev: loading out-of-tree module taints kernel.
[29899.080238] videodev: module verification failed: signature and/or required key missing - tainting kernel
[29899.080379] videodev: Unknown symbol mutex_lock (err 0)
[29899.080396] videodev: Unknown symbol __mutex_init (err 0)
[29899.080404] videodev: Unknown symbol mutex_lock_interruptible (err 0)
[29899.080417] videodev: Unknown symbol mutex_trylock (err 0)
[29899.080428] videodev: Unknown symbol mutex_unlock (err 0)
[29899.080444] videodev: Unknown symbol _raw_spin_unlock (err 0)
[29899.123271] Linux video capture interface: v2.00
```

Do you face the same error?

ev-mp, 2018-10-02:
Hello @aPonza , sorry for the intermittent response, the referred patch applies correctly to Ubuntu kernel as well as unmodified 4.8.15.
Looking deeper into the unresolved symbols it is probably due to differences in kernel configuration.
More specifically, the "#defines" are probably out of sync:
CONFIG_DEBUG_MUTEXES
CONFIG_UNINLINE_SPIN_UNLOCK

Can you provide the `.config` file that should be found in the root of the `/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/` folder ?
I'll try to use the RTPREEMPT patch to try and reproduce the exact error

aPonza, 2018-10-02:
Sure! [the_dot_config.txt](https://github.com/IntelRealSense/librealsense/files/2438408/the_dot_config.txt)

aPonza, 2018-10-03:
Thanks! Everything seems to work from a superficial test. The only errors/warnings were:
- during installation: 
<pre>
ERROR (dkms apport): kernel package linux-headers-4.8.15-rt10 is not supported
Error! Bad return status for module build on kernel: 4.8.15-rt10 (x86_64)
</pre>
makes sense, it's also stated in the installation guide and you reiterated it yourself;
<pre>
W: Possible missing firmware /lib/firmware/i915/kbl_guc_ver9_14.bin for module i915
W: Possible missing firmware /lib/firmware/i915/bxt_guc_ver8_7.bin for module i915
</pre>
the second one doesn't apply, but I do have a Kaby Lake processor... the firmware is [here](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915), and I installed following [this answer](https://askubuntu.com/a/832528/871401);
- while trying realsense-viewer:
<pre>
03/10 12:26:37,299 WARNING [139929276324416] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
</pre>
happens systematically at application startup, but it doesn't seem worth worrying over (?)

If everything seems normal to you as well, then this can be closed. Thanks for the help!

ev-mp, 2018-10-03:
It is not clear from the warning which dkms it refers to. anyhow, I'd suggest to remove librealsense dkms if installed since it is not verified to work with vanilla kernel in the first place, let alone real-time patch.

The rest seem normal.
You can attach a snapshot from the `realsense-viewer` with live stream and info overlay turned on (little (i) in viewport upper right) to verify that librealsense patches are in place

aPonza, 2018-10-03:
I still had the aptitude librealsense2-dkms installed from when I was trying the other kernel, maybe I had marked it as manually installed by mistake, now that's gone. This is a [screenshot](https://user-images.githubusercontent.com/39060879/46418919-023fee00-c72d-11e8-96fd-0eb0d4b9081e.png).

ev-mp, 2018-10-03:
The patches are in place. You can tell it by reading "HW Timestamp" parameter that confirms metadata retrieval. So far for integrating librealsense patches with a 3rd party patches.

The thing I find weird that in the attached image in place of RGB stream there was a snapshot of the viewer taken from a different POV with a recursive view :), though as if the RGB sensor was detached and physically moved outside the camera's case....

aPonza, 2018-10-04:
No no, the camera is all in one piece: I did say that is a screen-shot :) after all, the point of buying pre-built stereo cameras is also for their stable baseline, isn't it?

Thanks for the help in setting everything up, mr. Raikhel!

ev-mp, 2018-10-07:
Hello @aPonza , I'm glad everything is back on track.
Thanks for the feedback.

closed 2018-10-07 by ev-mp