realsenseai/librealsense #2396: Unity 2018.2.8 multiple bugs

TheBricktop, 2018-09-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (5.09.13) | 
| Operating System & Version |   {Win (10)}  | 
| Platform                                 | PC.  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C#/unity 2018.2.8 }                          | 
| Segment			|  {VR/AR }                          | 

### Issue Description
Downoaded Realsense SDK 2.0 for win10, installed everything, all drivers set, new project with imported intel realsense wrapper for unity package,
before even loading the main example scene im greeted with 38 bugs saying : 

>  Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs(95,171): error CS0433: The imported type `System.Action<T>' is defined multiple times

MartyG-RealSense, 2018-09-14:
If you are having problems with installation of the Unity wrapper, an easy setup shortcut is to drag and drop the pre-made DLL files in the RealSense SDK 2.0 folder into the Unity plugin folders of a new Unity project.  The link below describes the process.

https://communities.intel.com/message/525049#525049

matkatz, 2018-10-02:
There was a problem in the packaging process of the unity package.
The release page was updated with a new unity package both for 2.16.0 and 2.16.1.
You can also download the new package from [here](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.1/Intel.RealSense.unitypackage).

MartyG-RealSense, 2018-10-02:
Thanks a lot madkatz, I've posted a link to the update on the RealSense forum on the Intel Support site too!

closed 2018-10-16 by RealSense-Customer-Engineering