realsenseai/librealsense #2397: Adjust parameters in the spatial filter for best trade-off depth

Fangruiding, 2018-09-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10 64bits  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC|
| SDK Version                            |  {legacy / 2.11.0}                          | 
| Language                            |  {labview }                          | 
| Segment			| VR                         | 

### Issue Description
Hi I read the post-processing white paper about spatial filters. I made an experiment in Labview wrapper to find best filter's parameters. I put a squared box 2m in front of the camera firstly:
[image]

Then, I ran point cloud example in Labview and view the box in the following direction. I did so since I want to find whether the squared edges can be preserved:
[image]

Then, I modified alpha parameters to 0.3, 0.5 and 0.7 in sequence. I got these results:
When Alpha = 0.3, Delta = 8:
[image]

When Alpha = 0.5, Delta = 8:
[image]

When Alpha = 0.7, Delta = 8:
[image]

I then increased the Delta value without seeing obvious difference. But when the Delta drops below 6, the filter is kinda closed. So I sticked with 8.

Based on my observation, the alpha parameter should be 0.6 or 0.7 since I get the RMS error reduced while preserving the squared edges. When I turn it up to 0.8, the noise is way bigger. When I turned it down like 0.5, the squared edges are smoothed out. 

Do you guys think this test works for finding proper filter parameters at this distance?

Best,
Jake

agrunnet, 2018-09-14:
Yes that sounds like a very good approach.

Fangruiding, 2018-09-14:
Thanks for your prompt reply!

closed 2018-09-14 by Fangruiding