realsenseai/librealsense #2399: depth_frame filtering (post-processing)

ywjia, 2018-09-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   |5.10.3.0| 
| Operating System & Version |Ubuntu 16| 
| Kernel Version (Linux Only)    |4.15| 
| Platform                                 | PC|
| SDK Version                            |2.16.0| 
| Language                            |C++| 
| Segment			|VR| 

### Issue Description
For filtering (i.e., post-processing) the depth_frame, why does the following code give random results?

 rs2::spatial_filter spat_filter;
 rs2::temporal_filter temp_filter;
 rs2::disparity_transform depth_to_disparity(true);
 rs2::disparity_transform disparity_to_depth(true);
 depth_frame = depth_to_disparity.process(depth_frame);
 depth_frame = spat_filter.process(depth_frame);
 depth_frame = temp_filter.process(depth_frame);
 depth_frame = disparity_to_depth.process(depth_frame);

"original depth_frame"
[image]

"filtered depth_frame"
[image]

Thanks.

dorodnic, 2018-09-16:
Hi @ywjia 
```cpp
rs2::disparity_transform depth_to_disparity(true);
rs2::disparity_transform disparity_to_depth(true);
```
One of these needs to be `false`. 
Right now you are transforming to disparity twice.

ywjia, 2018-09-17:
Thanks, it works now.

closed 2018-09-17 by ywjia

AIBluefisher, 2018-11-29:
> Thanks, it works now.

Hi, @ywjia 
Can you show me your code ? Thanks so much!