realsenseai/librealsense #2400: How to execute depth_example or others in matlab?

shgim, 2018-09-15:
I did all the set-up for matlab use of the realsense wrapper.
but, I am not sure of how to execute or use the individual function or class of realsense matlab wrapper.
Can you let me know about this?

I built all the procedure as instruction describes, and set the path, 
I tried the following script in matlab command window, but all fail.

>> depth_example
Invalid character code sequence detected.
 
>> realsense.depth_example
Invalid character code sequence detected.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  matlab 2017a                         | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2018-09-17:
Hello @shgim, if you're building the wrapper from sources following [the guide](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#building-from-source), then you should have a directory called `+realsense` with mex and matlab wrapper.

Once you add this folder to Matlab search path, or change the current directory to be the location of `+realsense` this folder (see below), then you should be able to invoke the wrapper functions, e.g:
`realsense.depth_example`

[image]

pillaya21, 2018-09-23:
Hey, I added the folder to Matlab search path and also changed the directory to the folder, I still could not get the example to work. Could you help me with this, please? 
Camera Model: d435
Firmware: Latest
Matlab:  2018a
SDK version 2.0
windows 10 
[image]

pillaya21, 2018-09-26:
@ev-mp @dorodnic could any of you help me with this please?

ev-mp, 2018-09-26:
Hello @pillaya21 
The directory `+realsense`  (note the `+` in previous post ) should be in the current folder or the search path.

Switch current dir from `path\realsense` to `path` and rename `realsense` to `+realsense`.
Then run `realsense.depth_example`

shgim, 2018-09-29:
I cleared the above issue. 
the problem is caused from the folder name modification, + is very important to be remained in the realsense folder path. 
I appreciate the matlab wrapper for realsense, and also I am curious about how to make it.:)

closed 2018-09-29 by shgim

TDVL, 2019-02-21:
Hello shgim,
I have a similar problem. I would like to save a point cloud from D415 camera or from the bag file recorded with the RealSense Viewer.
Will you provide me the way of execution of Matlab wrapper functions: depth_example, pointcloud_example? What are the outputs? 
How can I extract the point cloud and RGB, XYZ arrays from "bag" file with Matlab, when I record a scene using RealSense Viewer in Win10? I am using RealSense D415&D435 and the data are stored in "bag" file.

I tried with matlab functions: rosbag, select, readMessages, but I can't extract the point cloud and the XYZ array of the scene/frame. I only got the RGB or the INFRA image data tih mentioned functions (the dimensions should be 640x480x3).

Or, how can I directly store the captured data as pointcloud1 message with XYZ and RGB arrays (307200x3 dimsnesions)?

I am using Win10 and Matlab. I need the point cloud data for image processing.

Can you help me? Also, I have the Matlab wrapper, but I can't extract the point cloud with the pointcloud_example function.

Thank you in advance.

shgim, 2019-02-26:
Dear users

On your question, actually I have not tried rosbag or ros message-related
function, I just tried to use basic api such as depth_example.
You can see detailed guideline to use realsense SDK using matlab on the
github web.
I just tried same and solved the problem by matching directory name (at
that time I missed + symbol in original folder, ex. +realsense)

Sorry for not helping you. and as I know, ros bag is another problem to be
dealt with  by MATLAB, and hope to you have success on it.
Thank you,

kind regards,
Suhyeon Gim.