realsenseai/librealsense #2402: Unable to install ROS package "ros-kinetic-librealsense" on Ubuntu 16.04.5

mikaelarguedas, 2018-09-17:
Unable to install the debian package `ros-kinetic-librealsense` on Ubuntu 16.04.5

Step to reproduce:

```
$ docker pull ros:kinetic
$ docker run -it --rm ros:kinetic
# apt update && apt upgrade -y
# apt install ros-kinetic-librealsense
```

Error message:
```
/var/lib/dpkg/info/ros-kinetic-librealsense.postinst: 40: /var/lib/dpkg/info/ros-kinetic-librealsense.postinst: udevadm: not found
libkmod: ERROR ../libkmod/libkmod.c:586 kmod_search_moddep: could not open moddep file '/lib/modules/4.15.0-34-generic/modules.dep.bin'
modinfo: ERROR: Module alias uvcvideo not found.
```

Additional information:
- Kernel version: 4.15.0-34-generic
- OS: Ubuntu 16.04.5 LTS

Installation [works on the ROS buildfarm](http://build.ros.org/view/Kdev/job/Kdev__realsense_camera__ubuntu_xenial_amd64/81) using:
- Kernel Version: 4.4.0-1060-aws
- Operating System: Ubuntu 16.04.4 LTS

ev-mp, 2018-09-17:
Hello @mikaelarguedas ,
The `ros-kinetic-librealsense`  is based on Librelasense1 and support v4.4 kernel.
In case you need to deploy it with kernel 4.15 I would suggest the following:
1. Switch to kernel 4.4 and then install the package normally, i.e:
`sudo apt-get install ros-kinetic-librealsense`.
2. Switch to kernel 4.15 and then [install](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) ` librealsense2-dkms` package only.

The dkms for librealsense2 is backward-compatible with all devices librealsense1 supported as well.

mikaelarguedas, 2018-09-17:
@ev-mp Thanks for the quick response!

I faced this issue as I wanted to simulate the turtlebot in Gazebo (I do not have access to realsense hardware). As `turtlebot_bringup` depends on [`realsense_camera`](https://github.com/turtlebot/turtlebot/blob/a66b6eafe008a4d150bf0ed9e490911cebcc3d8d/turtlebot_bringup/package.xml#L16) I am unable to install the `turtlebot_gazebo` simulation package.

Would it be possible to have a fallback behavior during the installation so that it prints a warning if the kernel version is not a supported one (but not fail to install) ? or is it preferable to make a request for the turtlebot packages to remove the optional dependency on `librealsense`?

@tfoote FYI

tfoote, 2018-09-17:
Looking at the LTS Kernel schedule https://wiki.ubuntu.com/Kernel/Support#LTS_Kernel_Support_Schedule

4.4 is the kernel version for 16.04.0 installations.
But 4.15 is the kernel for 16.04.5

According to @mikaelarguedas  It looks like we haven't started failing on the buildfarm as it's images are still running 16.04.04 which targets uses kernel 4.13 but I suspect the next time they get refreshed we'll switch to 16.04.05 and things will start breaking on the buildfarm too.

Likewise any new user who downloads and installs xenial now will get 16.04.05 by default. 

And as this is a core package installed by default in many metapackages including the TurtleBot we need to make sure that it's installable by default on our [primary supported platform Xenial](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021). If we cannot make this install by default and just work we'll have to strip the dependency which would be a pain and would significantly degrade the user experience for realsense users.

josephduchesne, 2018-09-18:
Can the postinst script be updated to check the kernel version and avoid this error on newer kernels?

tfoote, 2018-10-19:
It looks like there's a solution to 4.15 kernel support in #2498 how can we get this into the ROS repositories so that we can keep librealsense as part of the common installers and not fail to install on 16.04.5?

tfoote, 2018-10-23:
I just got a chance to reproduce this on a fresh 16.04.5 installation and got the following.

```
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20180809-140204-0800) ...

Creating symlink /var/lib/dkms/uvcvideo/1.1.1-3-realsense/source ->
                 /usr/src/uvcvideo-1.1.1-3-realsense

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Running the pre_build script:

dkms: Pre-build script started...

Downloading kernel sources...
Reading package lists...
E: You must put some 'source' URIs in your sources.list

ERROR: Could not download the required kernel sources to install RealSense camera support.

       To resolve, please follow the installation directions at:
           http://wiki.ros.org/librealsense#Installation_Prerequisites

Building module:
cleaning build area....(bad exit status: 2)
make KERNELRELEASE=4.15.0-38-generic -C /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/linux-src M=drivers/media/usb/uvc/....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.15.0-38-generic (x86_64)
Consult /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/make.log for more information.
WARNING: DKMS module failed to installed; removing...

------------------------------
Deleting module version: 1.1.1-3-realsense
completely from the DKMS tree.
------------------------------
Done.
WARNING: Unknown configuration for Intel RealSense(TM) cameras!
       To resolve, please follow the installation directions at:
           http://wiki.ros.org/librealsense#Installation
Processing triggers for libc-bin (2.23-0ubuntu10) ...
```

So it did fail to install but didn't block the installer and I could continue on to install the turtlebot packages on top.

sergii-strelkovskyi, 2018-10-31:
@mikaelarguedas Here is the reason why ros-kinetic-librealsense.postinst fails: systemd/systemd#8153

josephduchesne, 2018-11-01:
Some interesting output (18.04 host):

```
docker run -it --rm ubuntu:16.04 systemd-detect-virt
none
```

```
docker run -it --rm ubuntu:16.04 sh -c systemd-detect-virt
docker
```

```
docker run -it --rm ubuntu:16.04 bash -c systemd-detect-virt
none
```

(as mentioned by @lucaswerkmeister)
```
docker run -it --rm ubuntu:16.04 bash -c "systemd-detect-virt; true"
docker
```

KaleabTessera, 2019-02-19:
Any new on this?

sanjay-nag, 2019-06-24:
I'm facing this too, while installing ros-kinetic-turtlebot-stage! Any update on this?

`
Preparing to unpack .../ros-kinetic-turtlebot-stage_2.2.3-0xenial-20190608-162035-0800_amd64.deb ...
Unpacking ros-kinetic-turtlebot-stage (2.2.3-0xenial-20190608-162035-0800) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20190607-170900-0800) ...
modinfo: ERROR: Module uvcvideo not found.
dpkg: error processing package ros-kinetic-librealsense (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of ros-kinetic-realsense-camera:
 ros-kinetic-realsense-camera depends on ros-kinetic-librealsense; however:
  Package ros-kinetic-librealsense is not configured yet.

dpkg: error processing package ros-kinetic-realsense-camera (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of ros-kinetic-turtlebot-bringup:
 ros-kinetic-turtlebot-bringup depends on ros-kinetic-realsense-camera; however:
  Package ros-kinetic-realsense-camera is not configured yet.

dpkg: error processing package ros-kinetic-turtlebot-bringup (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of ros-kinetic-turtlebot-navigation:
 ros-kinetic-turtlebot-navigation depends on ros-kinetic-turtlebot-bringup; however:
  Package ros-kinetic-turtlebot-bringup is not configured yet.

No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                          No apport report written because the error message indicates its a followup error from a previous failure.
                                  dpkg: error processing package ros-kinetic-turtlebot-navigation (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of ros-kinetic-turtlebot-stage:No apport report written because MaxReports is reached already

 ros-kinetic-turtlebot-stage depends on ros-kinetic-turtlebot-bringup; however:
  Package ros-kinetic-turtlebot-bringup is not configured yet.
 ros-kinetic-turtlebot-stage depends on ros-kinetic-turtlebot-navigation; however:
  Package ros-kinetic-turtlebot-navigation is not configured yet.

dpkg: error processing package ros-kinetic-turtlebot-stage (--configure):
 dependency problems - leaving unconfigured
No apport report written because MaxReports is reached already
                                                              Processing triggers for libc-bin (2.23-0ubuntu10) ...
Errors were encountered while processing:
 ros-kinetic-librealsense
 ros-kinetic-realsense-camera
 ros-kinetic-turtlebot-bringup
 ros-kinetic-turtlebot-navigation
 ros-kinetic-turtlebot-stage
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

OS - Ubuntu 16.04.4 LTS
Kernel - 4.4.0-1060-aws

LucidOne, 2019-10-14:
I'm close to being able to put together a pull request.

## initrd
As the kernel module is not need to support filesystem access, I don't think it needs to be available at bootup so I would change `dkms.conf` to avoid docker/chroot initrd issues.
```
REMAKE_INITRD="no"
```

## Source
`E: You must put some 'source' URIs in your sources.list`
This is because `dkms-pre-build.sh` is running
```
apt-get source linux-image-${kernelver} | egrep -v '^ ' || true
```
so `/etc/apt/sources.list` needs something like
```
deb-src http://us.archive.ubuntu.com/ubuntu xenial-updates main
```
however since the GPG key isn't being verified
```
gpgv: Signature made Wed 18 Sep 2019 09:47:14 AM UTC using RSA key ID 70E1162B
gpgv: Can't check signature: public key not found
```
maybe the installer should [just use git](https://wiki.ubuntu.com/Kernel/SourceCode)
```
 git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial 
```
There can also be a permissions issue as `apt-get source` tried to drop permissions to download files. `chown _apt .` in the build script

## Signed Kernel
```
Picking 'linux-signed-hwe' as source package instead of 'linux-image-4.15.0-29-generic'
```
is because everything after 4.8.x uses a [signed kernel](https://packages.ubuntu.com/source/xenial/linux-signed-hwe) and `dkms-pre-build.sh` runs
```
apt-get source linux-image-${kernelver} | egrep -v '^ ' || true
```
and the "source package" for `linux-image-4.15.0-65-generic` is `linux-signed-hwe`
which doesn't actually contain the kernel source.

AFAIK needs to be replaced with something like
```
apt-get source linux-image-unsigned-${kernelver} | egrep -v '^ ' || true
```

## headers
linux-headers-4.15.0-64-generic
`Provides: linux-headers, linux-headers-3.0`
linux-headers-4.4.0-165
`Provides: linux-headers, linux-headers-3.0`
`linux-headers-generic` depends on `linux-headers-4.4.0-165-generic`

So the [package.xml](https://github.com/IntelRealSense/librealsense/blob/rosdebian/package.xml) should probably depend on `linux-headers`. Though it is probably also an Ubuntu upstream issue and `linux-headers-3.0` should probably be `linux-headers-4.0`.

## Format count is off
The current logic detects 5 formats, I think this is a typo with `"IR:Depth 26-bit (INZI)"` that was upstreamed in 4.15.0, I believe it should be removed but :man_shrugging: and I don't have hardware so I'm giving up here unless someone wants to help test. Perhaps @ev-mp can clarify as none of the patches seem to apply cleanly.

# Current Status
Confirmation of what repo/branch to make a pull request against would be helpful, I think it is [this one](https://github.com/IntelRealSense/librealsense/tree/rosdebian).

The closest to building I have gotten so far is
```bash
$ apt install linux-image-unsigned-4.15.0-65-generic
$ apt install linux-headers-generic dkms linux-headers-4.15.0-65-generic linux-headers-4.4.0-165-generic linux-headers-4.4.0-165 linux-headers-4.15.0-65
$ apt-get download ros-kinetic-librealsense
$ dpkg --unpack ros-kinetic-librealsense_1.12.1-0xenial-20190607-170900-0800_amd64.deb
$ patch /usr/src/uvcvideo-1.1.1-3-realsense/dkms-pre-build.sh << 'EOF'
39c39
< apt-get source linux-image-${kernelver} | egrep -v '^ ' || true
---
[... 49 lines cut ...]
   .fcc  = V4L2_PIX_FMT_RELI,
           ^
drivers/media/usb/uvc//uvc_driver.c:268:12: error: ‘UVC_GUID_FORMAT_L8’ undeclared here (not in a function)
   .guid  = UVC_GUID_FORMAT_L8,
            ^
drivers/media/usb/uvc//uvc_driver.c:269:11: error: ‘V4L2_PIX_FMT_Y8’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_Y8,
           ^
drivers/media/usb/uvc//uvc_driver.c:273:12: error: ‘UVC_GUID_FORMAT_L16’ undeclared here (not in a function)
   .guid  = UVC_GUID_FORMAT_L16,
            ^
drivers/media/usb/uvc//uvc_driver.c:278:12: error: ‘UVC_GUID_FORMAT_D16’ undeclared here (not in a function)
   .guid  = UVC_GUID_FORMAT_D16,
            ^
scripts/Makefile.build:330: recipe for target 'drivers/media/usb/uvc//uvc_driver.o' failed
make[1]: *** [drivers/media/usb/uvc//uvc_driver.o] Error 1
Makefile:1571: recipe for target '_module_drivers/media/usb/uvc/' failed
make: *** [_module_drivers/media/usb/uvc/] Error 2
make: Leaving directory '/var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/linux-hwe-4.15.0'
```

AFAIK the package.xml format doesn't have an option for conditional dependencies
so the generated debian package can't depend on `librealsense2-dkms | ros-kinetic-librealsense`

I tried to see if the dkms installer logic for `ros-kinetic-librealsense` could handle being installed after `librealsense2-dkms`. However I'm currently [blocked on that](https://github.com/IntelRealSense/librealsense/issues/4869).

LucidOne, 2019-10-22:
@RealSense-Customer-Engineering @RealSenseCustomerSupport Can we get an update on this?

ev-mp, 2019-12-01:
@LucidOne, @sanjay-nag  @tfoote , @KaleabTessera , mikaelarguedas  - 
The dkms package for Ubuntu Xenial has been [recently updated](https://github.com/IntelRealSense/librealsense/issues/5249) with regards to kernel 4.15.0-58+.
If you haven't updated the OS yet - please run `sudo apt update && sudo apt upgrade` to get the latest version of `librealsense2-dkms` and update accordingly

closed 2019-12-09 by ev-mp

mikaelarguedas, 2019-12-09:
Thanks @ev-mp for the follow-up

Great to know a fix is out! I since changed OS so didn't get a chance to test the fix

joana-ha, 2020-07-27:
I also have issues with installing ros-kinetic-librealsense on my ubuntu kernel 4.15.0-112-generic. I already installed librealsense2-dkms, which worked. However, when I want to install ros-kinetic-librealsense, it asks for patches:

```
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20191214-013909+0000) ...

Creating symlink /var/lib/dkms/uvcvideo/1.1.1-3-realsense/source ->
                 /usr/src/uvcvideo-1.1.1-3-realsense

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Running the pre_build script:

dkms: Pre-build script started...

Downloading kernel sources...
Reading package lists...
Picking 'linux-signed-hwe' as source package instead of 'linux-image-4.15.0-112-generic'
Need to get 13,8 kB of source archives.
Get:1 http://de.archive.ubuntu.com/ubuntu xenial-updates/main linux-signed-hwe 4.15.0-112.113~16.04.1 (dsc) [2.158 B]
Get:2 http://de.archive.ubuntu.com/ubuntu xenial-updates/main linux-signed-hwe 4.15.0-112.113~16.04.1 (tar) [11,7 kB]
gpgv: Signature made Fr 10 Jul 2020 06:32:57 CEST using RSA key ID 238F6A10
gpgv: Can't check signature: public key not found
dpkg-source: warning: failed to verify signature on ./linux-signed-hwe_4.15.0-112.113~16.04.1.dsc
dpkg-source: info: extracting linux-signed-hwe in linux-signed-hwe-4.15.0
dpkg-source: info: unpacking linux-signed-hwe_4.15.0-112.113~16.04.1.tar.xz
WFetched 13,8 kB in 0s (178 kB/s)
: Can't drop privileges for downloading as file 'linux-signed-hwe_4.15.0-112.113~16.04.1.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
grep: drivers/media/usb/uvc/uvc_driver.c: No such file or directory
INFO: No Intel RealSense(TM) cameras are currently supported.
Patching uvcvideo sources...
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/usb/uvc/Makefile b/drivers/media/usb/uvc/Makefile
|index c26d12fdb8f4..d86cf22155d1 100644
|--- a/drivers/media/usb/uvc/Makefile
|+++ b/drivers/media/usb/uvc/Makefile
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 14
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/usb/uvc/uvc_driver.c b/drivers/media/usb/uvc/uvc_driver.c
|index d11fd6ac2df0..914164137a13 100644
|--- a/drivers/media/usb/uvc/uvc_driver.c
|+++ b/drivers/media/usb/uvc/uvc_driver.c
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 100
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/usb/uvc/uvcvideo.h b/drivers/media/usb/uvc/uvcvideo.h
|index f0f2391e1b43..532ca19675a5 100644
|--- a/drivers/media/usb/uvc/uvcvideo.h
|+++ b/drivers/media/usb/uvc/uvcvideo.h
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 163
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/include/uapi/linux/videodev2.h b/include/uapi/linux/videodev2.h
|index a0e87d16b726..e6c1ec69cba4 100644
|--- a/include/uapi/linux/videodev2.h
|+++ b/include/uapi/linux/videodev2.h
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored

Building module:
cleaning build area....(bad exit status: 2)
make KERNELRELEASE=4.15.0-112-generic -C /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/linux-src M=drivers/media/usb/uvc/....(bad exit status: 2)
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/ros-kinetic-librealsense.0.crash'
Error! Bad return status for module build on kernel: 4.15.0-112-generic (x86_64)
Consult /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/make.log for more information.
WARNING: DKMS module failed to installed; removing...

------------------------------
Deleting module version: 1.1.1-3-realsense
completely from the DKMS tree.
------------------------------
Done.
WARNING: Unknown configuration for Intel RealSense(TM) cameras!
       To resolve, please follow the installation directions at:
           http://wiki.ros.org/librealsense#Installation
Processing triggers for libc-bin (2.23-0ubuntu11.2) ...
```
I found uvc_video.c and other files under `/usr/src/librealsense2-dkms-1.3.12/4.15.0/drivers/media/usb/uvc`, but obviously it's not found when I install ros-kinetic-librealsense. Also, when I ignore the patches during the installation, the ros camera nodelet can't be loaded:

`[FATAL] [1595855717.438950801]: Failed to load nodelet '/camera_sr300/disparity_registered_sw` of type `depth_image_proc/disparity` to manager `/camera/camera_nodelet_manager'
`
I need to install ros-kinetic-librealsense as the turtlebot ros packages depend on it. Can someone please help me with this issue?