realsenseai/librealsense #2404: Why the D435 extremely occupied cpu resources?

XJBDBB, 2018-09-17:
When I open the D435 in my demo,it will occupy the cpu resources unbearablely!! Only open camera and do nothing!!!
`rs2 pipeline pipe;`
`pipe.start()`
Somebaby can tell me what is wrong,I find lots documents also can not solve it.

#####Addtional
System Win10
Cpu   I7-8700K
Compiler Visual Studio 2017 community

ev-mp, 2018-09-17:
Hello @XJBDBB, can you provide some details on CPU utilization  ?

Calling `pipe.start()` configures and starts polling and processing data from device, i.e, polling, converting, synchronizing. The only step that is missing is rendering it.
The default configuration for D415 includes streaming both Depth and RGB stream at high resolutions and FPS, so processing those stream also includes color space conversion UYUV->rgb.

A decent reference to verify that your build is optimized is to download [Librealsense Windows installer](https://github.com/IntelRealSense/librealsense/releases) and run the same samples from it. Those samples are pre-compiled with optimizations. Comparing CPU utilization of these demos with your own code will provide you a benchmark.

XJBDBB, 2018-09-17:
@ev-mp Of course.In my demo,my CPU utilization is >90%.And I test lots manner,like using my default config `rs2::config cfg; cfg.disable_stream(RS2_STREAM_COLOR,-1);` but it also >90%.
And I run the exe from the samples,it only 10-20%.
Of course,I test Debug and Release,otherwise it still>90%.

RealSense-Customer-Engineering, 2018-09-18:
[Realsense Customer Engineering Team Comment]
Hi @XJBDBB,

What's your CMAKE build options? Did you disable "BUILD_WITH_OPENMP"?
Which librealsense version do you use?

XJBDBB, 2018-09-18:
@RealSense-Customer-Engineering @ev-mp Oh.Thank you for your reply.I have solved this problem which is my SDK version too old.I download the SDK2.14 and my CPU utilization is 10-30%.

closed 2018-09-18 by RealSense-Customer-Engineering