realsenseai/librealsense #2405: ROS turtlebolt3 interface for realsense not working

ShettyHarapanahalli, 2018-09-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry P  |
| SDK Version                            |  2                          | 
| Language                            |  C/python/ROS                          | 
| Segment			|  Robot/ROS                          | 

### Issue Description
Not able to launch the realsense with turtle3. 
I have used the command 
"roslaunch turtlebot3_bringup turtlebot3_realsense.launch" 

I have attached the log file and below is the Command prompt log - 
sshetty@sshetty-DL:~/github/turtlebot3/catkin_ws$ roslaunch turtlebot3_bringup turtlebot3_realsense.launch
... logging to /home/sshetty/.ros/log/7b2dd07a-ba7b-11e8-a346-54bf64471cc6/roslaunch-sshetty-DL-24987.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
[roslaunch-sshetty-DL-16202.log](https://github.com/IntelRealSense/librealsense/files/2388861/roslaunch-sshetty-DL-16202.log)

started roslaunch server http://sshetty-DL:38349/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/nodelet_manager-1]: started with pid [25004]
process[camera/driver-2]: started with pid [25005]
[ INFO] [1537192866.678508955]: Initializing nodelet with 12 worker threads.
[ERROR] [1537192866.834924318]: /camera/driver - No cameras detected!
[ INFO] [1537192866.835349170]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192871.846496628]: /camera/driver - No cameras detected!
[ INFO] [1537192871.846843128]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192876.853265135]: /camera/driver - No cameras detected!
[ INFO] [1537192876.853429339]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192881.863034063]: /camera/driver - No cameras detected!
[ INFO] [1537192881.863334031]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192886.873156554]: /camera/driver - No cameras detected!
[ INFO] [1537192886.873550006]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192891.877677120]: /camera/driver - No cameras detected!
[ INFO] [1537192891.877859514]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1537192896.887800951]: /camera/driver - No cameras detected!
[ INFO] [1537192896.888136182]: /camera/driver - Sleeping 5 seconds then retrying to connect
^C[camera/driver-2] killing on exit
[camera/nodelet_manager-1] killing on exit
[camera/nodelet_manager-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

ev-mp, 2018-09-17:
Hello @ShettyHarapanahalli ,
You've specified D435 as the unit under test, but the ros_launch was configured for R200 device
>/camera/driver/camera_type: R200

Another hint is
>/rosversion: 1.12.14 

that also corresponds to Librealsense1 version 1.12.

D435 is supported with Librealsense SDK v2 only. Please check and update

ShettyHarapanahalli, 2018-09-17:
When we are using the sdk the device abstraction will happen correct, will SDK 2 support old devices as well. I have a D435 and SR300 that needs to be interfaced on the same device/computer. 

Can you please share me the equivalent command for this in SDK 2.0 

"roslaunch realsense_camera r200_nodelet_rgbd.launch"

Is it
"roslaunch realsense2_camera rs_rgbd.launch"

closed 2018-09-24 by ShettyHarapanahalli