realsenseai/librealsense #2406: Program Crash (SIGSEGV) when calculating point cloud

ahoink, 2018-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 5.10.3.0 | 
| Operating System & Version |   Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.10.0-28                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.16.0                          | 
| Language                            |  C++                         | 

### Issue Description
I've been encountering some strange behavior. My program crashes when calling `pc.calculate(depth)` when built for Debug but seems to _mostly_ run fine when built for Release (Release will occasionally crash, Debug always crashes). 

Here is the call stack:

Thread [15] 2300 [core: 7] (Suspended : Signal : SIGSEGV:Segmentation fault)	
	rs2::pointcloud::calculate() at rs_processing.hpp:323 0x48195f	
	<lambda()>::operator()(void) const at main.cpp:1,372 0x47a6f9	
	std::_Bind_simple<main()::<lambda()>()>::_M_invoke<> at functional:1,531 0x48071e	
	std::_Bind_simple<main()::<lambda()>()>::operator() at functional:1,520 0x480674	
	std::thread::_Impl<std::_Bind_simple<main()::<lambda()>()> >::_M_run at thread:115 0x480604	
	std::this_thread::__sleep_for() at 0x7ffff135fc80	
	start_thread() at 0x7ffff18346ba	
	clone() at 0x7ffff0ac541d	

BEFORE updating the firmware and SDK , the crash occurred at the same function call, `pc.calculate(depth)`, but gdb gave a different call stack at crash:

Thread [14] 4952 [core: 2] (Suspended : Signal : SIGSEGV:Segmentation fault)	
	pthread_mutex_lock() at 0x7ffff19afd44	
	__gthread_mutex_lock() at gthr-default.h:748 0x4296fd	
	std::mutex::lock() at mutex:135 0x42a85c	
	std::unique_lock<std::mutex>::lock() at mutex:485 0x7ffff27a78a7	
	std::unique_lock<std::mutex>::unique_lock() at mutex:415 0x7ffff27a6b8a	
	single_consumer_queue<librealsense::frame_holder>::dequeue() at concurrency.h:51 0x7ffff27ccef8	
	rs2_wait_for_frame() at rs.cpp:891 0x7ffff29b6eaa	
	rs2::frame_queue::wait_for_frame() at rs_processing.hpp:171 0x41b4e8	
	rs2::pointcloud::calculate() at rs_processing.hpp:228 0x4816c2	
	<lambda()>::operator()(void) const at main.cpp:1,374 0x47a00d	

Code is part of a larger project but here's some relevant code:

	rs2::frameset data;
	rs2::align align(RS2:STREAM_COLOR);
	rs2::pointcloud pc;
	rs2::points points3d;
...

	if (pipe.poll_for_frames(&data))
	{
		rs2::frameset aligned_frames = align.process(data);
		color = aligned_frames.first(RS2_STREAM_COLOR);
		depth = (aligned_frames.get_depth_frame());
		points3d = pc.calculate(depth);
	}

ahoink, 2018-09-19:
Solved. Issue was unrelated to librealsense and caused by some kind of memory corruption from another library.

closed 2018-09-19 by ahoink

banuprathap, 2020-12-18:
@ahoink can you share a bit more info on this? I'm faced with the same problem and I'm trying to understand how you fixed it.

Did you have to remove/reinstall any libraries?