realsenseai/librealsense #2407: rs-measure gives incorrect measurements

ywjia, 2018-09-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.3.0| 
| Operating System & Version | Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.15                                   | 
| Platform                                 | PC|
| SDK Version                            |  2.16.0                       | 
| Language                            |  C++                          | 
| Segment			| VR| 

### Issue Description
I used the program "/librealsense-master/build/examples/measure" to take some measurements, but it seems that the measurements are way off. As shown below, the actual height of the night stand is 60cm but rs-measure gives a meausure of 48cm. 
[image]

Is there anything that need to be done about the camera, such as some kind of calibration?
Thanks.

ywjia, 2018-09-17:
From issue #2382, I know that "The de-projection function is deterministic and must use the intrinsic of the frame upon each it is invoked." But this is exactly how rs-measure.cpp did, as the related code segment below:

// Deproject from pixel to point in 3D
rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics(); // Calibration data
rs2_deproject_pixel_to_point(upoint, &intr, upixel, udist);
rs2_deproject_pixel_to_point(vpoint, &intr, vpixel, vdist);

Really puzzled.

ywjia, 2018-09-17:
On a second measure, I got 55cm, still way off the actual value of 60cm.

agrunnet, 2018-09-17:
Are you too close to object? There is a minimum distance of 45cm away when using d415 at 1280x720 resolution.

ywjia, 2018-09-18:
I had another observation that may be relevant to the above issue. Using "Depth Quality Tool", when the camera points at a flat wall 1500mm away, it shows "Distance" of ~1401mm.

agrunnet, 2018-09-18:
Sounds like a calibration issue then.

ywjia, 2018-09-29:
thanks. it works as expected after calibration.

closed 2018-09-29 by ywjia