realsenseai/librealsense #2408: How to save the color & depth streams into image in Unity with SDK 2.0 & D435 camera?

FernandoHo, 2018-09-18:
Hi there! Thanks for looking into my issue. 
The trouble I got is something related to the convert from texture to image. The "RsStreamTextureRenderer" script in Unity helps to update and bind the real-time stream to a RawImage conponent in the UI when a new frame is called with the texture.LoadRawTextureData() method. What I tried to do is save the texture to the image format. The texture was like this.
[image]
I used the EncodeToPNG() method to do the convert.

```
if(_stream.ToString()=="Depth" && frameCount%30==0)
File.WriteAllBytes("C:/Users/Ho Ho/Desktop/Gaze_robotic/Assets/ImageProcessing/" + "DepthImage" + Time.time.ToString() + ".png",texture.EncodeToPNG());
else if (_stream.ToString() == "Color" && frameCount%30 == 0)
File.WriteAllBytes("C:/Users/Ho Ho/Desktop/Gaze_robotic/Assets/ImageProcessing/" + "ColorImage" + Time.time.ToString() + ".png",texture.EncodeToPNG());
```
I could successfully get the color image, but the depth image turned out to be a "Filled with noise" one. I got really confused about this. Thanks for reading my question.
[image]
[image]

matkatz, 2018-09-26:
The depth image is 16 bits per pixel gray scale, is it possible that the viewer that you are using doesn't support this format?

FernandoHo, 2018-09-26:
Thanks for you reply Mat! The detail problem I found is that the depth data was stored in the format of "Texture.Format.R16" in Unity, which was described as "A 16 bit color texture format that only has a red channel", this texture format is only useful for native code plugins as there is no support for texture importing or pixel access for this format.  I think what I had to do is convert the 16 bits byte[] into 8 bits bytes[], and save them as a bmp file, but unfortunately I could not find a way out to achieve this. Do you have any idea about this? Or other ways to catch & save the depth stream in Unity with the SDK 2.0? Thanks again for looking into my issue!!

FernandoHo, 2018-09-26:
> The depth image is 16 bits per pixel gray scale, is it possible that the viewer that you are using doesn't support this format?

Thanks for you reply Mat! The detail problem I found is that the depth data was stored in the format of "Texture.Format.R16" in Unity, which was described as "A 16 bit color texture format that only has a red channel", this texture format is only useful for native code plugins as there is no support for texture importing or pixel access for this format. I think what I had to do is convert the 16 bits byte[] into 8 bits bytes[], and save them as a bmp file, but unfortunately I could not find a way out to achieve this. Do you have any idea about this? Or other ways to catch & save the depth stream in Unity with the SDK 2.0? Thanks again for looking into my issue!!

matkatz, 2018-09-26:
Depend on what you are trying to achieve, there are number of options:
1. Use RsProcessingPipe, the processing pipe has a colorizer filter that converts the depth image from 16 bpp gray scale (Z16) to RGB8 which you can save to disk in the same way you did with the color images.
The code should be something like this:
[image]
2. Use record/playback device:
[image]
This will save the the video streams in ROS bag file.
**Currently there is bug that cause Unity to crash at the end of the record, it will be fixed soon.**
 3. Use librealsense native tools, you can record/playback the video streams using [realsense viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.1/Intel.RealSense.Viewer.exe)
if you need to break the stream to frames you can than use [rs-convert tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert)

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi FernandoHo,

Did you try the methods suggested by matkatz below? Do you still need any help?

Thanks!

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
Hi FernandoHo,

We are going to close this one soon if there is nothing else.

Thanks!

closed 2018-10-30 by RealSense-Customer-Engineering