realsenseai/librealsense #2410: Installing D435 on Industrial PC with Ubuntu 16

Creesy87, 2018-09-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.09.09.02) | 
| Operating System & Version |   { Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<15>.<?> }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have installed the realsense and currently with an older firmware and SDK version since the newest versions seem not to work properly with ROS
I always had problems with Infrared 2 

[ WARN] [1537268734.312316279]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 480, FPS: 15

Now i used the rs-enumerate-devices command and got this result

 Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	817412071565
    Firmware Version              : 	05.09.09.02
    Recommended Firmware Version  : 	05.09.14.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/video4linux/video3
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	2.1

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 1	  1280x720	@ 4Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 1	  640x480	@ 4Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Infrared 1	  480x270	@ 4Hz	Y8
    Depth	  1280x720	@ 6Hz	Z16
    Depth	  1280x720	@ 4Hz	Z16
    Depth	  640x480	@ 30Hz	Z16
    Depth	  640x480	@ 15Hz	Z16
    Depth	  640x480	@ 6Hz	Z16
    Depth	  640x480	@ 4Hz	Z16
    Depth	  480x270	@ 60Hz	Z16
    Depth	  480x270	@ 30Hz	Z16
    Depth	  480x270	@ 15Hz	Z16
    Depth	  480x270	@ 6Hz	Z16
    Depth	  480x270	@ 4Hz	Z16

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color	  1280x720	@ 6Hz	RGB8
    Color	  1280x720	@ 6Hz	Y16
    Color	  1280x720	@ 6Hz	BGRA8
    Color	  1280x720	@ 6Hz	RGBA8
    Color	  1280x720	@ 6Hz	BGR8
    Color	  1280x720	@ 6Hz	YUYV
    Color	  640x480	@ 30Hz	RGB8
    Color	  640x480	@ 30Hz	Y16
    Color	  640x480	@ 30Hz	BGRA8
    Color	  640x480	@ 30Hz	RGBA8
    Color	  640x480	@ 30Hz	BGR8
    Color	  640x480	@ 30Hz	YUYV
    Color	  640x480	@ 15Hz	RGB8
    Color	  640x480	@ 15Hz	Y16
    Color	  640x480	@ 15Hz	BGRA8
    Color	  640x480	@ 15Hz	RGBA8
    Color	  640x480	@ 15Hz	BGR8
    Color	  640x480	@ 15Hz	YUYV
    Color	  640x480	@ 6Hz	RGB8
    Color	  640x480	@ 6Hz	Y16
    Color	  640x480	@ 6Hz	BGRA8
    Color	  640x480	@ 6Hz	RGBA8
    Color	  640x480	@ 6Hz	BGR8
    Color	  640x480	@ 6Hz	YUYV
    Color	  424x240	@ 60Hz	RGB8
    Color	  424x240	@ 60Hz	Y16
    Color	  424x240	@ 60Hz	BGRA8
    Color	  424x240	@ 60Hz	RGBA8
    Color	  424x240	@ 60Hz	BGR8
    Color	  424x240	@ 60Hz	YUYV
    Color	  424x240	@ 30Hz	RGB8
    Color	  424x240	@ 30Hz	Y16
    Color	  424x240	@ 30Hz	BGRA8
    Color	  424x240	@ 30Hz	RGBA8
    Color	  424x240	@ 30Hz	BGR8
    Color	  424x240	@ 30Hz	YUYV
    Color	  424x240	@ 15Hz	RGB8
    Color	  424x240	@ 15Hz	Y16
    Color	  424x240	@ 15Hz	BGRA8
    Color	  424x240	@ 15Hz	RGBA8
    Color	  424x240	@ 15Hz	BGR8
    Color	  424x240	@ 15Hz	YUYV
    Color	  424x240	@ 6Hz	RGB8
    Color	  424x240	@ 6Hz	Y16
    Color	  424x240	@ 6Hz	BGRA8
    Color	  424x240	@ 6Hz	RGBA8
    Color	  424x240	@ 6Hz	BGR8
    Color	  424x240	@ 6Hz	YUYV

ev-mp, 2018-09-18:
@Creesy87 , the camera is recognized as USB2 device:
>Usb Type Descriptor : 2.1

In USB2 mode only subset of streaming configurations is supported due to bandwidth limitation (IR2 is not).
Please check the cabling, you may also may try to switch USB port.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@creesy87 Could you please update if you still have issue after switch to USB3?

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
@creesy87 Any good news after switch to USB3? Please update. Thanks!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@creesy87 Could you please update? Thanks!

closed 2018-11-08 by RealSense-Customer-Engineering

fulkast, 2019-07-27:
Hi there,

I have a similar issue even in USB 3 mode. The output of `rs-enumerate-devices` is as follows:

Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	752112071099
    Firmware Version              : 	05.10.03.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	NO
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	3.2

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 1	  640x360	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 90Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  424x240	@ 90Hz	   Y8
    Infrared 1	  424x240	@ 60Hz	   Y8
    Infrared 1	  424x240	@ 30Hz	   Y8
    Infrared 1	  424x240	@ 15Hz	   Y8
    Infrared 1	  424x240	@ 6Hz	   Y8
    Depth	  1280x720	@ 30Hz	   Z16
    Depth	  1280x720	@ 15Hz	   Z16
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  848x480	@ 90Hz	   Z16
    Depth	  848x480	@ 60Hz	   Z16
    Depth	  848x480	@ 30Hz	   Z16
    Depth	  848x480	@ 15Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  640x480	@ 90Hz	   Z16
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
    Depth	  640x360	@ 6Hz	   Z16
    Depth	  480x270	@ 90Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  424x240	@ 90Hz	   Z16
[... 175 lines cut ...]
    Color	  320x240	@ 30Hz	   YUYV
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

arjunsengupta1998, 2020-02-18:
@fulkast I am having a similar issue with USB 3.0 
Infrared2 is not being displayed.
Were you able to solve the problem? Please let me know. Thanks

> Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	817412070030
    Firmware Version              : 	05.12.01.00
    Recommended Firmware Version  : 	05.12.01.00
    Physical Port                 : 	2-1.1-62
    Debug Op Code                 : 	15
    Advanced Mode                 : 	NO
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	822213020468
    Firmware Update Id            : 	822213020468

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 1	  640x360	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 90Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  424x240	@ 90Hz	   Y8
    Infrared 1	  424x240	@ 60Hz	   Y8
    Infrared 1	  424x240	@ 30Hz	   Y8
    Infrared 1	  424x240	@ 15Hz	   Y8
    Infrared 1	  424x240	@ 6Hz	   Y8
    Infrared 1	  256x144	@ 90Hz	   Y8
    Depth	  1280x720	@ 30Hz	   Z16
    Depth	  1280x720	@ 15Hz	   Z16
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  848x480	@ 90Hz	   Z16
    Depth	  848x480	@ 60Hz	   Z16
    Depth	  848x480	@ 30Hz	   Z16
    Depth	  848x480	@ 15Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  848x100	@ 100Hz	   Z16
    Depth	  640x480	@ 90Hz	   Z16
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
[... 184 lines cut ...]
    Color	  320x240	@ 30Hz	   YUYV
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

fulkast, 2020-02-18:
> @fulkast I am having a similar issue with USB 3.0
> Infrared2 is not being displayed.
> Were you able to solve the problem? Please let me know. Thanks
> 

Hey! I no longer use a realsense :slightly_frowning_face: but I remember that I fixed this by opening the realsense viewer GUI. There is a section in the options, somewhere on the top left to switch to professional mode. That brought up both infra channels

arjunsengupta1998, 2020-02-19:
@fulkast 
Yes, I did see that advanced mode was disabled on my device. It is working now. Thanks for your response! :)