realsenseai/librealsense #2411: I'm getting repeated frames when I try to read from a .bag file using pyrealsense2

alizarghami, 2018-09-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                            |   2.<?>.<?>  | 
| Language                            |  python  | 

### Issue Description

Hello realsense community

I'm facing an issue with reading frames from bag files which I recorded using a D415 realsense camera which I'll try to explain it brief.

I'm using the following code:

```
pipeline = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file(file_path)
profile = pipeline.start(cfg)

ts_list = []
fnum = 0

while True:

        # get frame from pipeline
    frames = pipeline.wait_for_frames() 
    
        # get frame timestamp
    ts = frames.get_timestamp()
         
        # get depth and color frames
    depth_frame = frames.get_depth_frame()
    color_frame = frames.get_color_frame()

        # Convert images to numpy arrays
    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
  
        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
    depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # create a depth matrix with size of rgb image
    depth_mx = depth2img(depth_frame)
        
        # define some tapes for patches 
    tape_y = [int(depth_mx.shape[0]/2) - 100, int(depth_mx.shape[0]/2) + 100]
    tape_x = [int(depth_mx.shape[1]/2) - 100, int(depth_mx.shape[1]/2) + 100]

        # define some patches to check existence of any object
    patch_center = depth_mx[tape_y[0]:tape_y[1], tape_x[0]:tape_x[1]]

        # define some conditions for frame selection
    cond = np.sum(patch_center) < 80000

        ## select and save frames
    if cond: ## center patch contains an object
        if not os.path.isdir('./selected_frames/'):
            os.mkdir('./selected_frames/')

        cv2.imwrite("./selected_frames/" + str(fnum) + "_" + str(ts) + "_rgb.jpg", color_image)
        np.save("./selected_frames/" + str(fnum) + "_depth_mx.npy", depth_mx)
        np.save("./selected_frames/" + str(fnum) + "_depth_colormap.npy", depth_colormap)
        print('frame num %d saved' % fnum)

    fnum += 1
```

The above code should read each frame from a .bag file (input_file) using pipeline, do some processing on the frame and save it if it meets some condition.

The issue is that my code gives me frames in wrong order. Also some frames are repeated many times in the output.

I can't figure out why is this happening and would be very thankful if someone can assist me with any solution.

RealSense-Customer-Engineering, 2018-09-20:
[Realsense Customer Engineering Team Comment]
Hi @alizarghami,

Without some processing there, just read the bag and save the files, is the frame order correct?

alizarghami, 2018-09-20:
> Without some processing there, just read the bag and save the files, is the frame order correct?
Thanks for answering me.
I have done such an experiment. Among 5000 first frames that I get from the pipeline (without any processing), I get about 700 unique timestamps. which means I received about 4300 repeated frames.

dorodnic, 2018-09-20:
This is a known issue, planned to be resolved in the upcoming release (#2360)

dorodnic, 2018-09-25:
Should be resolved now in [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

dorodnic, 2018-10-08:
Hopefully this can be resolved now
If you encounter more issues related to ROS recording, please open a new issue.

closed 2018-10-08 by dorodnic