realsenseai/librealsense #2412: D415 fails to stream when disable auto-exposure of RGB sensor

softyara, 2018-09-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.15.0-34-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.16.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
I developed a project with D415/435 under Qt5.10.1.
I read the color frame, depth and infrared frames. When I disable the auto-exposure, the sensor does not behave properly: it only starts the streaming on every second attempt. When it does not stream, this error appears:

> RealSense error calling rs2_pipeline_wait_for_frames(pipe:0xb630e0):
>     Frame didn't arrived within 5000

The key code segments for initialization and streaming:
```
...
  cfg.enable_stream(RS2_STREAM_INFRARED, m_depthWidth, m_depthHeight, RS2_FORMAT_Y8, 15);
  cfg.enable_stream(RS2_STREAM_COLOR, m_rgbWidth, m_rgbHeight, RS2_FORMAT_BGR8, 15);
  cfg.enable_stream(RS2_STREAM_DEPTH, m_depthWidth, m_depthHeight, RS2_FORMAT_Z16, 15);

  selection = pipe.start(cfg);

  realSenseSensor.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0); // color frames
  realSenseSensorDepth.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1); // depth frames

...

while(true)
{
  data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
  rs2::colorizer color_map;
  rs2::frame depth = color_map(data.get_depth_frame());
  rs2::frame color = data.get_color_frame();
}
...
```

In parallel I have the **realsense-viewer** opened, showing these messages in output section:

> Depth stream start failure
> IR stream start failure

One segment from the Realsense log-file:

>  18/09 17:10:39,931 DEBUG [140735901759232] (sync.cpp:516) fps 15 Depth 12425 2399906.851000 
>  18/09 17:10:39,931 DEBUG [140735901759232] (sync.cpp:529) (TS: I Depth I Infrared I Infrared I Color )Depth 12425 2399906.851000 fps 15 gap 66.666664 next_expected: 2399973.517664
>  18/09 17:10:39,931 DEBUG [140735901759232] (sync.cpp:516) fps 15 Depth 12425 2399906.851000 
>  18/09 17:10:39,931 DEBUG [140735901759232] (sync.cpp:516) fps 15 Depth 12425 2399906.851000 
>  18/09 17:10:39,931 DEBUG [140735901759232] (sync.cpp:334) (TS: I Depth I Infrared I Infrared I Color ) Depth 12425 2399906.851000  Wait for missing stream: 3 next expected 2399906.917664
>  18/09 17:10:39,933 WARNING [140735893366528] (backend-v4l2.cpp:887) Frames didn't arrived within 5 seconds
>  18/09 17:10:39,933 INFO [140735893366528] (types.h:926) Frames didn't arrived within 5 seconds
>  18/09 17:10:39,979 DEBUG [140735663965952] (sensor.cpp:444) FrameAccepted,Color,124250,12425,Arrived,1537283439915.121094 1537283439979.802002 diff - 64.680908 ,TS,2399907.451000,TS_Domain,Hardware Clock last_frame_number 12424 last_timestamp 2399840.251000
>  18/09 17:10:39,987 DEBUG [140735663965952] (archive.cpp:517) CallbackStarted,Color,12425,DispatchedAt,1537283439987.541748
>  18/09 17:10:39,987 DEBUG [140735663965952] (sync.cpp:143) DISPATCH (TS: )--> Color 12425 2399907.451000 
> 
>  18/09 17:10:39,987 DEBUG [140735663965952] (sync.cpp:143) DISPATCH (TS: I Depth I Infrared I Infrared I Color )--> Color 12425 2399907.451000 
> 
>  18/09 17:10:39,987 DEBUG [140735663965952] (sync.cpp:99) I Color--> Color 12425, 2399907.451000
> 
>  18/09 17:10:39,987 DEBUG [140735663965952] (sync.cpp:258) SYNC (TS: I Depth I Infrared I Infrared I Color )--> Color 12425 2399907.451000 

The **udevadm monitor** does not show any problems.

RealSense-Customer-Engineering, 2018-09-19:
[Realsense Customer Engineering Team Comment]
Hi @softyara,

I don't see this issue even though the color sensor's auto exposure is off. Do you see this issue using srealsense-viewer with your settings? what resolution do you use?

softyara, 2018-09-19:
> [Realsense Customer Engineering Team Comment]
> Hi @softyara,
> 
> I don't see this issue even though the color sensor's auto exposure is off. Do you see this issue using srealsense-viewer with your settings? what resolution do you use?

RGB cam is at 1920x1080, stereo module is at 1280x720, both at 15 fps. 

Actually, I do see similar problem in realsense-viewer with my settings: with auto-exposure disabled for color stream, whenever I turn on or turn off the RGB camera (while stereo module is constantly on), the stereo module seems to restart itself, sometimes throwing errors _'Depth stream start failure'_ and _'Incomplete frame detected! Size 279552 out of 1843455 bytes (15%)'_. The difference with my program with RealSense SDK is only that the realsense-viewer after the depth stream failure starts the stereo-module automatically anyway, whereas it just can't get any frames in my c++ program.

However, a couple of times I could reproduce the **exact** same problem with RGB sensor in realsense-viewer when it just could not start the Color stream, throwing two errors _'Frames didn't arrived within 5 seconds'_ and _'RT IC2 Config error'_ - in this case only complete restart of realsense-viewer helps resolving it.

There is the same behavior with both D415 and D435 modules. The D435 also outputs another error _'Right MIPI error'_.

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
Hi @softyara,

_These two errors 'RT IC2 Config error'_ and _'Right MIPI error' looks like the hardware issue. I wonder if unstable interconnection trouble there?_
_Are you able to check with the vendor to replace them?_

softyara, 2018-09-27:
> [Realsense Customer Engineering Team Comment]
> Hi @softyara,
> 
> _These two errors 'RT IC2 Config error'_ and _'Right MIPI error' looks like the hardware issue. I wonder if unstable interconnection trouble there?_
> _Are you able to check with the vendor to replace them?_

Indeed, seems like both of the cameras are faulty. Tried the third one today - works without issues.

justnick21, 2020-02-07:
We are experiencing the exact problem as described above. 

> When I disable the auto-exposure, the sensor does not behave properly: it only starts the streaming on every second attempt.

We have tested this across a number of cameras. Reliable recordings after we disabled auto-exposure was only observed in **2 of 13 cameras**.

Cameras experiencing this problem are from different batches, sometimes from different stores.

Our workaround for this at the moment is to hardware rest the cameras before setting params and initiating streams (or unplugging them and then plugging them back in).

justnick21, 2020-02-10:
@RealSense-Customer-Engineering could you re-open this ticket please?

MartyG-RealSense, 2020-02-15:
Apologies for the delay in responding.   I have re-opened the issue for you.

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@justnick21 Did you get any chance to try with the latest FW and SDK version to see if you still have this issue?

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@justnick21 We couldn't reproduce this issue with the latest FW and SDK here. Could you please try with the latest FW and SDK?

justnick21, 2020-03-02:
Hey, looking at this now, I will update shortly.

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@justnick21 Will close this if no other questions. Thanks!

closed 2020-03-23 by RealSenseCustomerSupport