realsenseai/librealsense #2413: poll_for_frames() failed to get a frameset

ywjia, 2018-09-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version              |5.10.3.0|
| Operating System & Version | Ubuntu 16| 
| Kernel Version (Linux Only)    |4.15  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.0| 
| Language                            |  C++  
| Segment			| VR                   | 

### Issue Description
I want to access frames in a bag file. To test that, I recorded a bag file "test_0000.bag" with color stream and depth steam. I can load and view it on realsense-viewer. However, the following code failed to get any frameset. Any ideas, please?

int main()
{
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file("test_0000.bag");
    cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 5);
    cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 5);
    rs2::pipeline_profile profile = pipe.start(cfg);
    rs2::frameset frameset;
    for(int i=0; i<30; i++) {
                if (pipe.poll_for_frames(&frameset)) {
                        std::cout << "got one frameset at " << i << std::endl;
                        break;
                }
                else
                        std::cout << "i = " << i << std::endl;
    }
    pipe.stop();
    return 0;
}

ywjia, 2018-09-18:
Note that the parameters in enable_stream() are consistent with the bag file. Even after removing these parameters, as below, still no frame was retrieved.

cfg.enable_stream(RS2_STREAM_COLOR);
cfg.enable_stream(RS2_STREAM_DEPTH);

Thanks for any help.

vladsterz, 2018-09-19:
_poll_for_frames()_ is an **non**-blocking function, so what you have achieved in your for-loop is iterate 30 times. If you want to get exactly 30 frames (or less than 30) you can try _wait_for_frames()_, which is blocking.

RealSense-Customer-Engineering, 2018-09-20:
[Realsense Customer Engineering Team Comment]
Hi @ywjia,

Your FPS is only 5 in your config. And, is your bag record also based on FPS=5? That means every 200ms, you can have the frames ready. Therefore, To iterate 30 times (based on your system clock) is not enough to have the frames. Suggest to use "poll_for_frames" in the loop is like below to make sure the frames available:

  rs2::frameset data;
  if (!pipe.poll_for_frames(&data)) {
   continue; // no frame available
  }
  rs2::frame depth = data.get_depth_frame();
  rs2::frame color = data.get_color_frame();

ywjia, 2018-09-20:
thanks for the clear explanation.

closed 2018-09-20 by ywjia