realsenseai/librealsense #2414: Install Intel RealSense viewer on windows 7

EmmanuelCos, 2018-09-19:
Hello,
I'm trying to install Inter RealSense viewer, I used the file Intel.RealSense.SDK.exe found on GitHub, version 2.16.0.
I'm using windows 7.
I manage to make the installation, but when I open Intel RealSense Viewer, it directly stops with an error.
Is the software not compatible with windows7?
Thank you,
Emmanuel

dorodnic, 2018-09-20:
Hello Emmanuel,
Technically, Windows 7 is not one of our supported platforms yet, as we state at the [release page](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.0). 
That said, we are doing work to enable this use case for the community, and we are very close (we are planning to include Win 7 installer in the upcoming release).

dorodnic, 2018-09-25:
Hi @EmmanuelCos 
Please take a look at [installation_win7.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md)

AaronZ101, 2018-09-25:
Does this not work for the SR300 though?

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
Hi @AaronZ101,

As you can see in the win7 installation guide, SR300 is not in the support list.

dorodnic, 2018-09-27:
I'll try to add it as soon as possible

closed 2018-10-01 by EmmanuelCos