realsenseai/librealsense #2415: Cannot build & install SDK2.0 from source on MacOS

threedlife, 2018-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  build2.16.0                           | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I have searched intensively and not yet found any instruction regarding building the SDK2.0 from source on MacOS directly. It did work with the Xcode project. I'm wondering if there is a way to build on terminal using cmake. 

I followed the instruction on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

```
mkdir build && cd build
sudo xcode-select --reset
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode
```

but when execute `make` it returns an error 

> make: *** No targets specified and no makefile found.  Stop.

So my questions are

1. how to build the SDK from source and install it in /usr/local, not using the xcode project?
2. how to build examples?

dorodnic, 2018-09-20:
If you are specifying `-G Xcode` then the next step should not be `make` but rather `Open and build the XCode project` as in the document. If you remove `-G Xcode` then you should be able to run `make`

dorodnic, 2018-09-25:
Hi @threedlife 
Did you try without `-G` or with XCode?

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
HI @threedlife,

Are you able to use "make" and successfully build & install librealsense according to @dorodnic's comments?

gaurav945, 2018-09-28:
Hey, so, adding to the query, I am trying to install Pyrealsense2 on Mac. As mentioned in this issue : https://github.com/IntelRealSense/librealsense/issues/2118, I need to build a python wrapper, using the instructions on https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#building-from-source & then use the mac installation given on https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_osx.md
Now, my question is, should I do the python wrapper thing after the Xcode building or before ? And also, once I execute this command : cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true, the next command after this, "make -j4", gives an error saying, makefile not found. What do I do ? I just want pyrealsense2 on my mac, please help. :\ :\

dorodnic, 2018-09-28:
You can add `-DBUILD_PYTHON_BINDINGS=bool:true` to `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode` and then build with XCode or you can run `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true` (without `-G`) and then `make -j4`. Both will work, one using the IDE and the other via the command line. 
Please note that CMake "remembers" variables for the directory, so either delete the contents of `build` if you are switching between methods (`rm -rf ./*`) or use different folder instead of `build` (`build_clang` and `build_xcode`)

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Hi @threedlife, @gaurav945,

Still need any support for this topic?

closed 2018-10-28 by dorodnic

0a01-wait, 2020-02-19:
Hi.  Having (almost) the same problem as OP.  I followed the instructions at https://github.com/IntelRealSense/librealsense/blob/8e7db477a69741ca413a6b4d6402d19c8aace00e/doc/installation_osx.md and am getting the error 
The source directory "/users/eric/build" does not appear to contain CMakeLists.txt.
I am trying to setup for use with Xcode 11.3.1, Realsense 435.  TIA