realsenseai/librealsense #2418: Calculate point cloud in matlab

schmiran, 2018-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.0                        | 
| Language                            | matlab 2018b                          | 
| Segment			|  others                    | 

### Issue Description
Hello everyone, I'm trying to calculate a point cloud in Matlab. The depth_example works fine. But if I now try to calculate the point cloud from the depth values, I don't get any data back. What am I doing wrong?

```matlab
pipe = realsense.pipeline();
pcl_obj = realsense.pointcloud();
profile = pipe.start();
for i = 1:5
        fs = pipe.wait_for_frames();
end
pipe.stop();
depth = fs.get_depth_frame();
points = pcl_obj.calculate(depth);
data = points.get_data();
```

I then get the warning message: "Warning: Can't detect frame dims, sending only first pixel "
First pixel is always 0. I expected something like a matrix with the x, y, z values of the points.

Thank you very much for your help!

schmiran, 2018-09-21:
Thank you for the answer. Okay, then I'll probably implement it myself. I thought, because there is a pointcloud.m with the function calculate in the wrapper, that it might already be supported.

MartyG-RealSense, 2018-09-21:
I have seen the point cloud script in the MATLAB wrapper too.  I do not have any details about the workings of the wrapper.  The presence of that script makes me think that the wrapper may support point clouds but an example tutorial that shows how to access it has not been published yet.

schmiran, 2018-09-21:
My mistake! I had apparently changed something in points.m. I get my point cloud now. On ``points`` simply execute ``get_vertices()`` and you have your x, y, z values. Thanks anyway for the help! Greetings

MartyG-RealSense, 2018-09-21:
Great, I'm very glad you found a solution.  :)

closed 2018-09-21 by schmiran

TDVL, 2019-02-16:
Hello schmiran,
I saw your issue about pointcloud. I am a new RealSense camera user an I use Matlab and the RealSense Viewer for recording. The recorded file is in bag format. Now, I need the pointcloud, the RGB, INFRA, XYZ data in MAtlab. Can I extract this in Matlab from the bag file? If I understood you succeeded. Can you help with your script?