realsenseai/librealsense #2420: Black color image in current Master

BGA3, 2018-09-20:
Hi

With the current software on the master branch, I received all black color images. This is both using the Realsense viewer and my custom c++ script. And both using camera firmware 5.10.3 and 5.9. 

Can someone please fix this?

RealSense-Customer-Engineering, 2018-09-20:
[Realsense Customer Engineering Team Comment]
Hi @BGA3,

Which platform/board do you use? What's your librealsense version?
I previously saw the similar issue on Odroid XU4 board.

BGA3, 2018-09-20:
Hi,

I am working on a Firefly roc-rk3328-cc board running Ubuntu 16.04. The librealsense version is the current master.

dorodnic, 2018-09-20:
Yes, we are working on it - #2306
The issue is limited to non-Intel boards

dorodnic, 2018-09-25:
Thank you!
Should be resolved in [2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

BGA3, 2018-09-26:
Thanks for the support. It seems to be fixed, at least it works on my setup.

closed 2018-09-26 by BGA3