realsenseai/librealsense #2424: Recording multiple clips consecutively

upxela, 2018-09-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.14.0                           | 
| Language                            |  python2.7   |

### Issue Description
I am looking to recording many shorter, consecutive videos (~2 minutes in length) rather than one large video (which may amount to 24+ hours). Therefore it is desirable that there is not much lost between videos. Unfortunately my code that between closing a pipeline and starting the pipeline again, it takes approximately 0.6 seconds, which is not optimal for my application. Does anyone have any suggestions for workarounds for this?

upxela, 2018-09-20:
@MartyG-RealSense Thanks for the reply! Do you know how I would use this in tandem with the realsense api, specifically the pipeline start/close commands?

MartyG-RealSense, 2018-09-20:
My experience with ROS is limited and one of the Intel team can give better advice on it.  You should be able to use the SDK's ROS wrapper though.

https://github.com/intel-ros/realsense/releases

upxela, 2018-09-20:
It looks like this method requires Linux. Are you aware of any Windows-friendly methods to obtain the same result?

MartyG-RealSense, 2018-09-20:
There is a Windows version of the newest ROS 2, though it likely would not be able to talk to the SDk API through the ROS wrapper.

https://github.com/ros2/ros2/wiki/Windows-Install-Binary

ROS2 can be made compatible with the 400 Series cameras.

https://communities.intel.com/message/543051#543051

MartyG-RealSense, 2018-09-20:
An easier solution may be to try the developer UnaNancyOwen's recording program for SDK 2.0 to see if that gives you smaller bag file sizes.

https://github.com/UnaNancyOwen/RealSense2Sample/tree/master/sample/Record

upxela, 2018-09-21:
Thanks for the suggestions @MartyG-RealSense. As of right now, file size is not the main problem; the point of recording multiple shorter films is such that in the case of a file corruption, there is not as much of an impact if only 2 minutes are lost compared to 24 hours.

Additionally, I have seen problems with frame drops after 30 minutes when streaming full depth resolution with 4 cameras concurrently, which brings about the idea of using shorter clips to prevent as big of an impact should it happen during an important recording session.

upxela, 2018-09-21:
Unfortunately, each camera in my application would be part of its own "system," i.e. camera 1 would not record anything that camera 2 is recording, etc. Therefore it is important for me to obtain a seamless video from each camera.

MartyG-RealSense, 2018-09-21:
It's a tricky situation you're dealing with.  The SDK has a function called Keep(), where you can store frames in memory until the end of streaming and then save them, and do some post-processing on the frames too if you wish (though this adds to processing time).  This way, you may lose less frames than if you are recording the frames as you go. 

Keep() is best suited to short recording sequences, so it may be a good fit for how you are recording the data in multiple chunks/  

Another option may be to use an industrial-grade USB hub and / or industrial-grade USB cable, though these are more expensive then regular USB equipment.  Industrial grade equipment is used in applications such as medical scanners, where frame drops cannot be tolerated.

upxela, 2018-10-09:
@MartyG-RealSense frame dropping is currently not as big of an issue as just being able to record consecutive videos. I realized that to close a pipeline and restart can take up to approximately 1.5 seconds (~0.5 seconds to start, ~1 second to close), which is too long of a pause for my application.

Is there a way that I can redirect the file destination of my camera recording without closing the pipeline? I know that you previously mentioned using ROS, but I can't find any documentation that explains how to use it in tandem with realsense.

MartyG-RealSense, 2018-10-10:
I carefully went over the code of the SDK's record and playback sample program.  

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

Whilst it may be possible to change the path with code by having the filename as a number (e.g 1.bag) and incrementing it by 1 with a command, I could not see a way to start another file without first stopping the pipeline.  It is the stopping of the pipeline that releases currently held resources, including the bag file that was being recorded.

One of the Intel guys may be able to offer further suggestions.

upxela, 2018-10-10:
Thanks for the insight @MartyG-RealSense! Hopefully someone from Intel can find a workaround.

closed 2018-12-12 by RealSense-Customer-Engineering