realsenseai/librealsense #2425: what is the right way of setting depth camera parameters such as "high accuracy mode"?

ywjia, 2018-09-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435| 
| Firmware Version                   | 5.10.3.0| 
| Operating System & Version |   Ubuntu 16| 
| Kernel Version (Linux Only)    |  4.15|
| Platform                                 | PC|
| SDK Version                            |  2.16.0| 
| Language                            |  C++ | 
| Segment			|  VR| 

### Issue Description
I know that we can use set_option to set parameters for the depth sensor, as below

depth_sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.0001);

what is the right way of setting the mode of the depth camera, such as "high accuracy" or "high density" mode?

Thanks.

MartyG-RealSense, 2018-09-20:
The discussion linked to below may help you with setting Visual Presets with C++ code.

https://communities.intel.com/message/540135#540135

closed 2018-10-02 by ywjia

dibya-pati, 2018-10-22:
Hi,
I have been trying to do this using the pyrealsense2 wrapper, but so far failed. Would appreciate any lead.
Can't find any documentation for python

MartyG-RealSense, 2018-10-23:
The only example of a Visual Preset script for Python I could find was one provided by Dorodnic in November 2017.

https://github.com/IntelRealSense/librealsense/issues/856#issuecomment-347828788

fburakk, 2019-10-07:
> 
> 
> The discussion linked to below may help you with setting Visual Presets with C++ code.
> 
> https://communities.intel.com/message/540135#540135

Hi,

The link doesn't seem to work. Is there any source that you can give about this topic?

Thanks

MartyG-RealSense, 2019-10-07:
Some of the links on the Intel Support forum got broken when they moved to a new forum system.  The new link to that discussion is below.  It doesn't seem to be helpful to your needs to load a json with C++ though - I just include it here so others can find it.

https://forums.intel.com/s/question/0D50P0000490SEVSA2/how-to-apply-visual-presets-in-c

 This link here is likely to be more useful for loading a json with C++:

https://github.com/IntelRealSense/librealsense/issues/3037#issuecomment-452732168