realsenseai/librealsense #2427: realsense-viewer: Cannot access /sys/class/video4linux

elzKit, 2018-09-21:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.10 | 
| Operating System & Version |  Ubuntu 16| 
| Kernel Version (Linux Only)    |            4.15.0-30-generic                             | 
| Platform                                 | PC.  |
| SDK Version                            | 2.0                      | 
| Language                            |                         | 
| Segment			|                       | 

After installation of SDK from packages according to 
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
realsense-viewer does not connect to the camera and prints the following error to the console
 (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux

Installed packaged are:
librealsense2/xenial,now 2.16.0-0~realsense0.85 amd64 [installed,automatic]
librealsense2-dkms/xenial,xenial,now 1.3.1-0ubuntu3 all [installed]
librealsense2-udev-rules/xenial,now 2.16.0-0~realsense0.85 amd64 [installed,automatic]
librealsense2-utils/xenial,now 2.16.0-0~realsense0.85 amd64 [installed]

The installation seemed successful, but the following error was present
Loading the modified modules into kernel...modprobe: ERROR: could not insert 'hid_sensor_gyro_3d': Unknown symbol in module, or unknown parameter (see dmesg)
dmesg log is attached

[dmesg_log.txt](https://github.com/IntelRealSense/librealsense/files/2404942/dmesg_log.txt)

RealSense-Customer-Engineering, 2018-09-24:
[Realsense Customer Engineering Team Comment]
Hi @elzkit,

Please run "lsmod | grep video" to see if video module is loaded correctly.

ev-mp, 2018-09-25:
Hello @elzKit ,
According to the kernel log the patched module was compiled with a different version, which doesn't make sense. Is it possible
It is also not clear from the report what the following line means:
> ~~librealsense2-dkms/xenial,xenial,now 1.3.1-0ubuntu3 all [installed]~~

Does it mean that `librealsense2-dkms` package was/could not be installed?
What is the current output of `dkms status`?

If the package was not properly installed, then having the log for `sudo apt install librealsense2-dkms` is essential to assess what went wrong. Additionally, please run `dmesg -wT` simultaneously in another window to monitor and collect kernel logs.

elzKit, 2018-10-01:
I did a completely fresh install of the system and the issue is gone. Was certainly due to some mix up of kernel versions.

closed 2018-10-01 by elzKit