realsenseai/librealsense #2428: Error on `try_to_fetch_usb_device()`

mprat, 2018-09-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10 | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                      | 
| Realsense version | 2.16.0 |
| Platform                                 | NVIDIA Jetson  |
| Language                            |  C++                        | 

### Issue Description

I am using a D415 camera from a TX2 Jetson board. When enabling advanced mode and loading a JSON file, I consistently get this error:

```
 20/09 17:40:09,786 DEBUG [547734253568] (context.cpp:100) Librealsense VERSION: 2.16.0
 20/09 17:40:09,945 DEBUG [547734253568] (ds5-factory.cpp:467) try_fetch_usb_device(...) failed.
 20/09 17:40:15,869 WARNING [547734253568] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
RealSense error calling rs2_load_json(dev:0x5010e0, json_content:0x50b150, content_size:2834):
    get_xu(id=11) failed! Last Error: Input/output error
```

The solution seems to be to reboot the TX2 device (just unplugging and replugging in the camera does not resolve the error).

A similar issue (#2043) suggests doing a hardware reset using `dev.hardware_reset()`, but it doesn't work. I think it's a problem with the USB bus not resetting properly when closing the camera stream. After a system reboot I can start the stream maybe once or twice before seeing this error again.

* What is the proper way to "cleanly" close a camera?
* What is the right way to reset your USB bus?
* What is another way to avoid this error?

RealSense-Customer-Engineering, 2018-09-28:
[Realsense Customer Engineering Team Comment]
Hi @mprat,

I don't see this issue when I tried the realsense-viewer of librealsense 2.16.1 to load a json file. Your D415 firmware is 5.10.3, right?
What's your Jetson TX2 BSP version? Is it L4T  28.2.x? Also, suggest to upgrade to the latest BSP to resolve some stability issues.

mprat, 2018-09-28:
D415 Firmware 5.10.3, Jetson TX2 L4T 28.2.x (latest one), librealsense 2.16.0 - should I be switching to 2.16.1?

Regardless of these issues, what is the cleanest way to close a camera? Is it enough to release the pipeline? Or is it something more?

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @mprat,

See #1086 to combine the hardware reset with `rs2::device_hub` to have a synchronous flow.
Also, suggest to upgrade to the latest firmware and librealsense to see if the failure still there.

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Hi @mprat,

Still need any support for this topic? If not, will close this.

closed 2018-12-04 by mprat