realsenseai/librealsense #2431: How to install library realsense on Nvidia Jetson Xavior

duclinhfetel, 2018-09-24:
Hi guys,
I use cmake & make to build source librealsense
When I built 59%, received log error
"
_[ 59%] Built target pyrealsense2
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by 'examples/software-device/rs-software-device'.  Stop.
CMakeFiles/Makefile2:395: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2_
"
Can you help me problem?
Thank for all to support.

dorodnic, 2018-09-24:
Hi @duclinhfetel 
There seems to be some missing dependency related to OpenGL (for the graphical examples).
I'd try:
```sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev```
And if one of the packages cannot be installed, try:
```./scripts/install_glfw3.sh``` (from the repo root directory)

If all else fails, you can always add `-DBUILD_GRAPHICAL_EXAMPLES=false` to your cmake and continue without the OpenGL stuff.

umersadiq, 2018-10-19:
Does anyone have a working Xavier with librealsense?

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@duclinhfetel Did you try what Dorodnic suggested? Please update when you get chance to try. Thanks!

ghormann, 2018-10-19:
Similar problem on my Xavier.    The packages listed by Dorodnic are installed.

libglu1-mesa-dev is already the newest version (9.0.0-2.1build1).
libglfw3-dev is already the newest version (3.2.1-1).
libgl1-mesa-dev is already the newest version (18.0.5-0ubuntu0~18.04.1).

[ 59%] Built target pyrealsense2
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by 'examples/software-device/rs-s                                                                oftware-device'.  Stop.
CMakeFiles/Makefile2:395: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
nvidia@jetson-0423318029403:~/librealsense/build$ ls /usr/lib/aarch64-linux-gnu/libGL.so

dorodnic, 2018-10-22:
Hi @duclinhfetel @ghormann 
If you manage to figure out this problem please consider contributing your notes to [installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
We don't yet have a Xavier board to test with.

Tien94, 2018-10-23:
Hi,
I've successfully installed librealsense2 in Xavier with Jetpack4.1 as followed setups:

1. Install dependencies:
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

2. Build source: cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3

3. Except one issue that i have to run export PYTHONPATH=$PYTHONPATH:/usr/local/lib before the application, librealsense2 seems working fine with Xavier so far. I've add this path to bashrc file but this did not work.

Hope this helps.

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
@Duclinhfete Could you please try as Tien94 provided? Looking forward to your update. Thanks!

ghormann, 2018-10-24:
Tien94's solutions is similar to the first post.  (Skip the examples. ).  This is more of a workaround than a solution.   (Still don't think it is a problem IntelRealSense, but rather a dependency.)

ghormann, 2018-10-25:
Today there is now an update to libGL in the standard repos that fixed the library linking issue.  After an apt-get update and apt-get upgrade, I'm able to compile when using -DBUILD_EXAMPLES=true

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@duclinhfetel Is the ticket resolved? Looking forward to your reply. Thanks!

duclinhfetel, 2018-11-07:
@dorodnic  @umersadiq  @RealSense-Customer-Engineering  @ghormann 
Thank for all support.
My problem is resolved

closed 2018-11-07 by RealSense-Customer-Engineering

atyshka, 2018-11-21:
@duclinhfetel I'm ordering a Xavier and want to use the realsense with it. Were you able to just install librealsense without having to do the kernel patches needed on the TX2 or will I still need to do those?

ghormann, 2018-11-21:
Installed fine on our Xavier without patching kernel.

pavloblindnology, 2018-11-28:
What about patching? The latest release still doesn't support kernel 4.9.

atyshka, 2018-11-28:
Yeah the kernel needs patching but I’m not sure how to go about it for a Xavier

sequae92, 2018-11-30:
@ghormann Could you please elaborate the steps in installing librealsense on Xavier ? As @pavloblindnology  mentioned, there is no support for the 4.9 kernel which Xavier has. So how did you go about the installation?

atyshka, 2018-11-30:
@sequae92 just follow the steps from Tien94. They work just fine for me

sequae92, 2018-11-30:
Thanks for the reply.
I am having a trouble when I do make install after cmake.

This is the error I receive
In file included from /home/nvidia/workspace/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pytypes.h:12:0,
                 from /home/nvidia/workspace/librealsense/wrappers/python/third_party/pybind11/include/pybind11/cast.h:13,
                 from /home/nvidia/workspace/librealsense/wrappers/python/third_party/pybind11/include/pybind11/attr.h:13,
                 from /home/nvidia/workspace/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pybind11.h:43,
                 from /home/nvidia/workspace/librealsense/wrappers/python/pybackend.cpp:4:
/home/nvidia/workspace/librealsense/wrappers/python/third_party/pybind11/include/pybind11/detail/common.h:111:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.

Would you know how to proceed ?

ghormann, 2018-11-30:
Looks like you are missing a dependency.

sudo apt-get install python-dev   # for python2.x installs
sudo apt-get install python3-dev  # for python3.x installs

MartyG-RealSense, 2019-02-19:
Though this case is closed, I wanted to add for future readers that the JetsonHacks website published a guide in January 2019 to installing Librealsense2 on the Jetson AGX Xavier.

https://www.jetsonhacks.com/2019/01/21/intel-realsense-d435i-on-nvidia-jetson-agx-xavier/

dookei, 2020-05-20:
sudo apt-get --reinstall install libglvnd-dev solve it for me

MartyG-RealSense, 2020-05-20:
@dookei   Thanks for your advice!    :)