realsenseai/librealsense #2433: Recording of IR stream

stekla79, 2018-09-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win (8.1/10)  | 
| SDK Version                            | 2.0                          | 
| Language                            |  {C++ }                          | 
| Segment			|  Others                         | 

Hi, 
I have more of a general question regarding the realsense D435, C++ and OpenCV. I'm building an application that grabs the frames from the realsense camera, process the depth stream in OpenCV which acts on kind of a motion detection and size algoritm to track moving objects in the scene and if the pass the center line of the camera. 

I would like to record a video of each object and starting 5 seconds before it passes the center line, and also preferably the IR stream since this has a greater field of view than the RGB camera. At the moment I am creating a buffer with a vector of 300 openCV mat objects creating a 10 second video with 5 second data before and after crossing the line with the help of timers. But when I try to write these Mat's to disk I block the program since it is a blocking operation. And I can't have the program blocked.  

So my question is what do you think would be the best way to avoid the blocking or is there another smarter way to circumvent this issue? 

1. Should i put the writing of the video through openCV in a separate thread? I haven't used threads before but I will give it a shot if you think it would work. 

2. Or could I access the IR stream directly through the Realsense SDK and write the video without openCV, is this possible? 

3. Or if I could access the IR stream through a middle layer as FFMPEG this would also work well I think but I can only see the RGB stream through VLC, not the IR streams. Is it possible to access the IR streams without the SDK?

dorodnic, 2018-09-24:
Hi @stekla79 
It's an interesting question. 

We provide built-in recorder that uses ROS-bag file format. You can use this recorder in parallel to your main application (just enable it via `config.enable_record_to_file()`). You can also control when to start / stop recording using `pause` / `resume` methods on the recorder object (cast the device you get from `pipeline_profile` to `recorder` using `as<recorder>()`)
When using the built-in recorder, we handle threading for you, but not compression (due to potential licensing conflicts with FFMPEG). 

If you chose to build something custom, please note that you will probably need to use `frame.keep()` to store frames. If I recall correctly, `cv::Mat` object will not actually copy the pixel data, but will only store the pointer. We have some examples on how to send frames for processing on a background thread in [the wiki](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#do-processing-on-a-background-thread)

stekla79, 2018-09-25:
Thank you for your reply. ROS-bag files gets really large and I'm only interested in a video of the object triggered the passage of the centerline. 

I am aware this is not a specific realsense question, more of a non blocking programming problem. Unfortunately I don't have a lot of experience with C++ programming and threading, which I believe could solve the issue. I made some blunt attempts with threading without success, it is still blocking the program. I guess I have to learn more.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@stekla79  For recording method, Dorodnic has provided the detailed explanation. Do you still have any other question about this ticket?

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@stekla79 Do you still have question about this ticket? Looking forward your update. Thanks!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@stekla79 Could you please update? Thanks!

RealSense-Customer-Engineering, 2018-11-08:
[Realsense Customer Engineering Team Comment]
@stekla79  Considering the ticket resolved, close it accordingly. Thanks!

closed 2018-11-08 by RealSense-Customer-Engineering