realsenseai/librealsense #2436: RGB at Half Framerate

upxela, 2018-09-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.13.0 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC.  |
| SDK Version                            |  2.14                          | 
| Language                            |  python2.7                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When I record RGB video on D435 at 30fps for 2 minutes and check the number of color image messages in the rosbag, I see I have ~1800, when I expect double the amount (2 min * 60 sec/min * 30fps = 3600 frames). I cranked the fps to 60, which to my surprise raised the number of messages to around 3600, when I would be expecting double the amount. 

I'm using pyrealsense2 from the PyPI distributionfor python2.7. Has anyone experienced the same or a similar phenomenon?

ev-mp, 2018-09-25:
Hello @upxela ,
A common reason for FPS drop is the sensor's embedded auto-exposure feature.
You can check the actual FPS  in realsense-viewer and then and play with auto-exposure and manual exposure values to find a proper configuration.
There is also "Auto-Exposure Priority" control to prioritize FPS over exposure in low-light conditions.
Also see #1957

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@upxela Could you please try below two options to see if you still get the issue?

1. Disable "Auto-Exposure" and set it manually, verifying it does not exceed the upper bound.
2. Leave "Auto-Exposure switched on, but disable "Auto-Exposure Priority" control to enforce constant FPS.

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
@upxela Could you please provide the update of this issue? Thanks!

closed 2018-11-04 by dorodnic