realsenseai/librealsense #2438: Viewer feature requests

gpred, 2018-09-25:
The realsense viewer would really benefit from these features. It would be great if they could be added,

Viewer feature requests,
- Playback slider to show milliseconds
- Hovering the mouse over image in 2D and 3D mode should show x, x and z coordinates of closest point to 4 decimal places
When playing back a recorded .bag file,
- Ability to export all frames to a specified folder (or a specified time range)
- This export option should allow the specification to choose what to export, eg. depth images, .ply, .pcd etc
- The timestamp of all frames should be exported to the same folder, either appended to each file name or in a separate readable file which identifies the timestamp of each output file
- An output file which records all the camera settings that were used during the capture
- The ability to skip forward and back one frame (this is currently disabled)

AntoineWefit, 2018-10-01:
I would like to ad a request : could it be possible to make the "options" panel resizable ?

Right now, there are some options (the ones with a drop-down list in particular) with which you really can't see the selected values, even when extended. Larger drop-down lists or a larger panel may help correcting it, but those are just ideas...

**Even when the list is extended, it is not possible to see some values** (_Post-Processing/Temporal Filter/Stream Format Filter_ for example)
[image]

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
Pending on JIRA DSO-10648

RealSenseCustomerSupport, 2020-05-09:
----------------------------------------------

@gpred The latest RealSenseViewer has implemented some features you suggested such as below. For other features, our engineers are still evaluating and working on. Thanks!

- The timestamp of all frames should be exported to the same folder, either appended to each file name or in a separate readable file which identifies the timestamp of each output file
- The ability to skip forward and back one frame

RealSenseSupport, 2021-04-06:
@gpred Some update up to now.

- The feature of "Playback slider to show milliseconds" has already been implemented in SDK v2.43. 
- When playing back a recorded .bag file,

1. Ability to export all frames to a specified folder (or a specified time range) 
2. This export option should allow the specification to choose what to export, eg. depth images, .ply, .pcd etc

The above two features are already implemented in a stand-alone `rs-convert` tool 
https://github.com/IntelRealSense/librealsense/tree/master/tools/convert 

- output file which records all the camera settings that were used during the capture. 
Many frame parameters are already available in the frame's metadata which is part of the record - frame size, format, the controls values.

RealSenseSupport, 2021-04-25:
@gpred Most of the requested features have been implemented. Do you have any other questions? Can we close this? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-05-10:
@gpred Any other questions about this? Please note that this ticket will be closed if we don't hear from your for another 7 days. Thanks!

RealSenseSupport, 2021-05-18:
@gpred No update from you for weeks. As most of the features have been implemented, close this accordingly. Thanks!

closed 2021-05-18 by RealSenseSupport