realsenseai/librealsense #2443: 3D reconstruction from point cloud to triangular mesh c++

EmmanuelCos, 2018-09-25:
Hello,
I'm using the D435 camera on windows. I would like to know if there exists opensource code to make 3D surface reconstruction from the point cloud images given by RealSense Viewer, that is to say, convert the point cloud into a mesh of triangles? I need ideally to have that code in C++.
Thank you,
Emmanuel

AntoineWefit, 2018-09-28:
I know this solution is not a C++ code, but it is open source : 
You could use [Meshlab](http://www.meshlab.net/), which is great to manipulate point clouds and meshes. Beside doing usually such processing by hand, it is also possible to use scripts to transform your data with Meshlab Server. This means that you can export your ".ply" from the RealSense Viewer and then, execute a simple batch file to transform and process your data to a point cloud. Let me know if you are interested.

Hope this will help.

dorodnic, 2018-09-28:
There is some work being done on exporting PLY faces (#2036), integration with [Open3D](http://www.open3d.org/) and [assimp](http://www.assimp.org/). Also there is a very basic [example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl) for working with PCL and cross-platform installation guide that you might find useful. I think PCL offers a lot of what you are looking for, but it is starting to show its age.

RealSense-Customer-Engineering, 2018-09-28:
[Realsense Customer Engineering Team Comment]
@EmmanuelCos

You can convert .bag files recorded from RealSense Viewer with the "rs-convert" tool in LibRealSense.

In order to convert .bag files into a .ply file (3d mesh file):

1)  Record a .bag file in RealSense Viewer.

2)  Use rs-convert with these commands:

rs-convert -i some.bag -l some_dir/sample.ply

There will be a list of .ply files associated with each each recorded frame.

3)  At that point you can use MeshLab to view and manipulate generated .ply files.

Please let us know if your question has been answered,

Thank you!

closed 2018-10-01 by EmmanuelCos