realsenseai/librealsense #2448: How to build librealsense2 on 4.17 TK1

jalcim, 2018-09-26:
HI,

Im work on bsp on a board tegra124 (tk1).
We are a stable kernel 4.17 for this platform.
Now we need build librealsense2 for armhf 32bit.

I use this line for that :
ARCH=armhf MACHINE=arm-linux-gnueabihf cmake ../ -DBUILD_EXAMPLES=false

first error : 
arm-mep-linux-gnueabi-gcc: error: unrecognized command line option '-mssse3'

then i supress arg '-mssse3' here : 
build/third-party/realsense-file/CMakeFiles/realsense-file.dir/flags.make
build/CMakeFiles/realsense2.dir/flags.make
build/CMakeFiles/realsense2.dir/link.txt
build/compile_commands.json

As expected I have a lot of warning.

At 27% of build i have an error
librealsense/src/media/ros/ros_writer.h:300:56: error: 'hexfloat' is not a member of 'std'
             frame_timestamp_msg.value = to_string() << std::hexfloat << pose->get_frame_timestamp();

Then my question.
Can we build librealsense2 for armhf 32bit ? (armv7)
For jetson TK1 (kernel 4.17)
I have read a lot of documents on jetsonhack before comming write here.

Thanks

jalcim, 2018-09-26:
Ok i have find.

On "CMakeLists.txt" line 762
`    if(${MACHINE} MATCHES "arm-linux-gnueabihf")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
    elseif(${MACHINE} MATCHES "aarch64-linux-gnu")
        set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mstrict-align -ftree-vectorize")
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mstrict-align -ftree-vectorize")
    else()
        set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mssse3")
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mssse3")
        set(LRS_TRY_USE_AVX true)
    endif(${MACHINE} MATCHES "arm-linux-gnueabihf")`

Its because on the yocto sdk the name of the compilator differ 
then when i specify (MACHINE=arm-linux-gnueabihf) for the cmake.
its not recognised because the cmake prefer see environment,
then the if doesn't match and the else is executed.

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
Hi @jalcim,

It's good you found the root cause. If you resolve the issue, suggest you can create a PR to help improve the build. Check below for the detail.
https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md

closed 2018-10-15 by RealSense-Customer-Engineering

sidmangla, 2018-10-17:
Hi there I receive the same error. Can you explain the steps taken to install realsense on tegra tk1

RealSense-Customer-Engineering, 2018-10-18:
[Realsense Customer Engineering Team Comment]
Hi @sidmangla,

You can check below link to see how it works on your TX1.
https://github.com/jetsonhacks/buildLibrealsense2TX

sidmangla, 2018-10-18:
Hi there,

Thank you for fast response. But I have a tk1 chip that to on toradex
carrier board.

Specs:
Ubuntu:14.04
Kernel: 3.10.40
Cuda:6.5

Thank you
Siddharth Mangla

sidmangla, 2018-10-18:
Hi there,

I am stuck at issue #2071
Undefined reference to symbol 'XConvertSelection' #2071. Please help I
don't know what seems to be the problem.
I have getting the same error. I have Ubuntu 14.04 and kernel -3.10.40 on
toradex apalis tk1. I have cmake: 3.12
G++- 5.5
Gcc -5.5
Glfw-3.1.2

Thank you
Siddharth Mangla

jalcim, 2018-10-19:
gcc-4.8 not implemented fully C++11 then you need to patch gcc-4.8 with : 
https://gcc.gnu.org/ml/gcc-patches/2014-03/msg01505.html

Information for tx1 is totaly useless.
.
IMPORTANT linux 3.10 don't have uvc1.5 only uvc1.1 then you need to use libuvc,
for that go on the CMakeLists.txt and at line 660 : 
replace set(BACKEND RS2_USE_V4l2_BACKEND)
by set(FORCE_LIBUVC_ON)

sidmangla, 2018-10-19:
Hi there,
Thanks for quick reply.
I will try the above mentioned steps. Once I build it how will I be able to
get pyrealsense2 on 14.04 as the repos for the demo and tools are for 16.04
and 18.04 and not for 14.04. Pip for pyrealsense2 is also not available for
14.04 (I checked).What can be done to use the sdk through python on 14.04
Ubuntu.

Thank you
Siddharth Mangla

jalcim, 2018-10-22:
I don't know i use C/C++.
If you want to use python wrapper you need to hack it.
good luck.

sidmangla, 2018-10-22:
Could you please help?

monsieurmona, 2019-05-05:
I am not sure if anyone is interested in this after that long time. I own a Intel RealSense Depth Camera D435 and try to get it running on a Jetson TK1.

I took the repository https://github.com/jetsonhacks/buildLibrealsense2TX as a basis and solved all the compile errors for
* kernel
* librealsense

You may find the patches and scripts in 
https://github.com/monsieurmona/buildLibrealsense2TK1

jawad-tech, 2019-07-26:
Hi i have also faced this issue "hexfloat is not a meber of std" 
I solved this issue by 
1) Removing and purging the gcc 4.9.4 which was installed on my system
2) sudo add-apt-repository ppa:ubuntu-toolchain-r/test
3)  sudo apt-get update
4) sudo apt-get install gcc-5 g++-5
5) sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5
6) sudo update-alternatives --set gcc "/usr/bin/gcc-5

Or you follow the first step from here and then step 2,3,4,5,6 from 
[image]

jalcim, 2019-07-26:
1 TK1 kernel is 3.10.40 and realsense need 4.4 for uvc 1.5 driver
2 the nvidia driver for gk20a (the gpu) is only coded for 3.10.40

then you can't switch for 4.4 or more recent version.
and the backport of uvc is realy hard.
the only easy solution is to use "libusb" but your performance died instantly !