realsenseai/librealsense #2450: Guide to compiling IMGUI libraries?

HippoEug, 2018-09-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

### Issue Description
Hello,

Does anyone know how to import the Imgui library into a C++ Project? I have tried many methods but none seem to work.

I am trying to get an example [Streams Alignment](librealsense/examples/align at master · IntelRealSense/librealsense · GitHub ) to work, but no matter what i try, #include <imgui.h> for example still does not work even though I have added it in Solution Explorer itself etc.

Apologies for such a question, this is challenging stuff for me. Only managed to figure out how to create a new project from scratch with the Intel Realsense D435 after many trials and error.

Having difficulty using CMAKE as well. Have never used it before...

Would appreciate any guidance. Thank you once again

dorodnic, 2018-09-27:
Hi @HippoEug 
You need to add couple of files to your project:
* imgui.cpp
* imgui_draw.cpp
* imgui_impl_glfw.cpp
In addition, if you are using Visual Studio, under project properties, C/C++ / General, add IMGUI directory to the Additional Include Dirs (for me `C:\git\librealsense\examples\align\..\..\third-party\imgui`)
In `main`, include:
* `#include <imgui.h>`
* `#include "imgui_impl_glfw.h"`
This should work, but let us know if you encounter any problem

HippoEug, 2018-09-27:
Hello Dorodnic, thank you for taking your time to reply. Firstly, I must apologize for being a complete noob.

Here is my Project File:

[image]
1. Are the 3 files you mentioned (imgui.cpp, imgui_draw.cpp, imgui_impl_glfw.cpp) enough? Do I also need to copy the imgui.h files or imgui_impl_glfw.h files anywhere?

Here is the Windows File Explorer of where the the third party imgui files are for me:

[image]
2. Do I reference this in the Additional Include Directories?

[image]
3. If so, how would you type it in here?

Currently here is how my IDE looks:
[image]
4. Even #include "../example.hpp" is an error. I only have #include "example.hpp". I've never been so lost...

Seriously, thank you in advance for taking the time to help me out. I would buy you a coffee if i could

dorodnic, 2018-09-27:
Sorry to hear, getting samples compilation streamlined for Windows is on our radar for some time now, but we can't manage to get to it. 

In step 1/2, yes, just add the path to imgui directory in the Additional Include Directories window. It will be easier then copying individual *.h files to your project, although you could also do that. Click on the white space in the dialog to add folder. 

The 3 cpp files you need to add to your project from Visual Studio (Source Files -> Add -> Existing Item)
IMGUI is not "header only", so the easiest solution is to compile it as part of your project. This is what we do in the SDK.

This should work, but I personally think that the best way to get started with the SDK is downloading [the sources from GitHub](https://github.com/IntelRealSense/librealsense/archive/v2.16.1.zip) and [building everything using CMake](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md). 
This way you have access to everything and its pretty much guarantied to work.

HippoEug, 2018-09-27:
Somehow I got the CMake to work. First time using CMake. 

I plan on further understanding the examples and libraries and make a custom application using the Realsense. Do you suggest I modify the program from the Generated Cmake program?
[image]

Also, would love it if you guys could fix the samples code without using CMake. And also more in depth tutorials for noobs like myself. 

Thank you once again Sergey, you have no idea how much I appreciate your generous help :)

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@hippoeug Any other questions about this ticket?  Please update. Thanks!

HippoEug, 2018-10-19:
No more issues, other issue is in another ticket! Please close this, thanks! 👍

closed 2018-10-19 by RealSense-Customer-Engineering