realsenseai/librealsense #2451: Calibration tool source code

SirDifferential, 2018-09-27:
I'm reading through this: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400%20_Custom_Calib_Paper.pdf

The calibration tool seems to be only included as a Windows executable. We'd like to run it on Linux with a differently sized checkerboard pattern. Is the tool open source and can the code be downloaded somewhere? I found the non-checkerboard calibration here: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_Programmer.pdf, but we'd prefer using a checkerboard.

dorodnic, 2018-09-27:
I think there is a Linux version of the tool. @RealSense-Customer-Engineering could you please help with the link?
The code is unfortunately not available as open-source.

SirDifferential, 2018-09-27:
Ah well. We'll have to print your checkerboard pattern and use it then. If you ever revisit the program, maybe add the checkerboard dimensions as a command line parameter or something.

MartyG-RealSense, 2018-09-27:
Intel has a document with a proper checkerboard for calibration that can be printed off:

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

closed 2018-10-04 by SirDifferential