realsenseai/librealsense #2452: Can't Build Examples for opencv

lelouedec, 2018-09-27:
Good afternoon, 
I am trying to make librealsense work with Opencv and wanted to try the examples given in wrappers. I followed the instructions, and built successfully Opencv, did the cmake as told and when comes the time to make I keep having the following errors for the opencv examples. 

> [ 55%] Building CXX object wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/rs-imshow.cpp.o
> In file included from /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/rs-dnn.cpp:7:0:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/../cv-helpers.hpp: In function ‘cv::Mat frame_to_mat(const rs2::frame&)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/../cv-helpers.hpp:27:24: error: ‘CV_RGB2BGR’ was not declared in this scope
>          cvtColor(r, r, CV_RGB2BGR);
>                         ^
> In file included from /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/latency-detector.h:9:0,
>                  from /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/rs-latency-tool.cpp:4:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/../cv-helpers.hpp: In function ‘cv::Mat frame_to_mat(const rs2::frame&)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/../cv-helpers.hpp:27:24: error: ‘CV_RGB2BGR’ was not declared in this scope
>          cvtColor(r, r, CV_RGB2BGR);
>                         ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/rs-dnn.cpp: In function ‘int main(int, char**)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/rs-dnn.cpp:56:41: error: ‘cvGetWindowHandle’ was not declared in this scope
>      while (cvGetWindowHandle(window_name))
>                                          ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/dnn/rs-dnn.cpp:127:44: error: ‘CV_FILLED’ was not declared in this scope
>                      Scalar(255, 255, 255), CV_FILLED);
>                                             ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/imshow/rs-imshow.cpp: In function ‘int main(int, char**)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/imshow/rs-imshow.cpp:22:59: error: ‘cvGetWindowHandle’ was not declared in this scope
>      while (waitKey(1) < 0 && cvGetWindowHandle(window_name))
>                                                            ^
> In file included from /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/rs-grabcuts.cpp:6:0:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/../cv-helpers.hpp: In function ‘cv::Mat frame_to_mat(const rs2::frame&)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/../cv-helpers.hpp:27:24: error: ‘CV_RGB2BGR’ was not declared in this scope
>          cvtColor(r, r, CV_RGB2BGR);
>                         ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/rs-grabcuts.cpp: In function ‘int main(int, char**)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/rs-grabcuts.cpp:48:59: error: ‘cvGetWindowHandle’ was not declared in this scope
>      while (waitKey(1) < 0 && cvGetWindowHandle(window_name))
>                                                            ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/grabcuts/rs-grabcuts.cpp:64:30: error: ‘CV_BGR2GRAY’ was not declared in this scope
>          cvtColor(near, near, CV_BGR2GRAY);
>                               ^
> In file included from /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/rs-latency-tool.cpp:4:0:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/latency-detector.h: In constructor ‘detector::detector(int, int)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/latency-detector.h:159:45: error: ‘CV_FONT_HERSHEY_SIMPLEX’ was not declared in this scope
>              Point(display_w / 2 - 470, 30), CV_FONT_HERSHEY_SIMPLEX,
>                                              ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/latency-detector.h: In member function ‘void detector::update_instructions()’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/latency-detector.h:273:28: error: ‘CV_FONT_HERSHEY_SIMPLEX’ was not declared in this scope
>              Point(80, 20), CV_FONT_HERSHEY_SIMPLEX,
>                             ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/rs-latency-tool.cpp: In function ‘int main(int, char**)’:
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/rs-latency-tool.cpp:46:30: error: ‘CV_WINDOW_NORMAL’ was not declared in this scope
>      namedWindow(window_name, CV_WINDOW_NORMAL);
>                               ^
> /home/lelouedec/Desktop/librealsense/wrappers/opencv/latency-tool/rs-latency-tool.cpp:58:41: error: ‘cvGetWindowHandle’ was not declared in this scope
>      while (cvGetWindowHandle(window_name))
>                                          ^
> wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/build.make:62: recipe for target 'wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/rs-imshow.cpp.o' failed
> make[2]: *** [wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/rs-imshow.cpp.o] Error 1
> CMakeFiles/Makefile2:254: recipe for target 'wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/all' failed
> make[1]: *** [wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/build.make:62: recipe for target 'wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o' failed
> make[2]: *** [wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o] Error 1
> CMakeFiles/Makefile2:419: recipe for target 'wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all' failed
> make[1]: *** [wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all] Error 2
> wrappers/opencv/grabcuts/CMakeFiles/rs-grabcuts.dir/build.make:62: recipe for target 'wrappers/opencv/grabcuts/CMakeFiles/rs-grabcuts.dir/rs-grabcuts.cpp.o' failed
> make[2]: *** [wrappers/opencv/grabcuts/CMakeFiles/rs-grabcuts.dir/rs-grabcuts.cpp.o] Error 1
> CMakeFiles/Makefile2:309: recipe for target 'wrappers/opencv/grabcuts/CMakeFiles/rs-grabcuts.dir/all' failed
> make[1]: *** [wrappers/opencv/grabcuts/CMakeFiles/rs-grabcuts.dir/all] Error 2
> wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/build.make:62: recipe for target 'wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/rs-latency-tool.cpp.o' failed
> make[2]: *** [wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/rs-latency-tool.cpp.o] Error 1
> CMakeFiles/Makefile2:364: recipe for target 'wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/all' failed
> make[1]: *** [wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/all] Error 2
> Makefile:127: recipe for target 'all' failed
> 

OpenCv is working fine with any other C++ code using it.
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.10.03.00) | 
| Operating System & Version |   {Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.15.0-34-generic        | 
| Platform                                 | PC NVIDIA .  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {c++ }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-09-27:
> RealSense examples have been desinged and tested with OpenCV 3.4, Working with latest OpenCV 4 requires minor code changes
From [opencv.md](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv)

Please try with opencv 3.4. 
Obviously you can work with RealSense and any version of opencv, however, just the included examples have trouble building with the newer versions.

lelouedec, 2018-09-28:
> Please try with opencv 3.4.
> Obviously you can work with RealSense and any version of opencv, however, just the included examples have trouble building with the newer versions.

Good afternoon,
Thank you for your answer, I did install opencv3.4 on my computer as mentioned in the repository. 
So I guess my problem doesn't come from there.

dorodnic, 2018-09-28:
Perhaps another version of opencv is installed? You can print the CMake variables of librealsense using `cmake -LA` in the build directory, perhaps it can help us troubleshoot. 
I'm jumping to this direction because the symbols and methods in error messages are exactly the ones that got changes between 3.4 and 4 opencv versions. 
Please share any findings and we will do our best to help

lelouedec, 2018-09-28:
Ok now it does work perfectly, thank you for your help it is a solved issue :)

dorodnic, 2018-10-08:
Awesome! You are welcomed to share more feedback on any problems you encounter

closed 2018-10-08 by dorodnic