realsenseai/librealsense #2455: Align color to depth

alt01, 2018-09-28:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Realsense D435 depth camera} |  
| Operating System & Version |   {Win 10}  | 
| Platform                                 | PC |
| Language                            | {matlab, python } | 

I have some pairs of depth and color images aquiered from a d435 Realsense depth camera , and I want to align the color image to the depth image.

I have the intrinsics and extrinsics values of the camera i used.

Do you know how can I acomplish this?

MartyG-RealSense, 2018-09-28:
If you want to align color with depth using Python, a guide published by Intel may be useful to you.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

To clarify:
You have your own extrinsic between your own color camera (not D435 color camera) and D435 depth camera?

alt01, 2018-10-01:
No, I am using the D435 depth and color camera, and I have the intrinsics and extrinsics values:

Depth intrinsics: width: 424, height: 240, ppx: 214.063, ppy: 120.214, fx: 213.177, fy: 213.177, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]

Color intrinsics: width: 424, height: 240, ppx: 214.031, ppy: 117.083, fx: 307.266, fy: 307.39,model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]

depth to color extrinsics: rotation: [0.999966, 0.00585115, -0.00579111, -0.00586071, 0.999982, -0.00163449, 0.00578144, 0.00166838, 0.999982]
 translation: [0.0146822, 0.0002266, -3.92719e-05]

-  color to depth extrinsics: rotation: [0.999966, -0.00586071, 0.00578144, 0.00585115, 0.999982, 0.00166838, -0.00579111, -0.00163449, 0.999982]
 translation: [-0.0146833, -0.000140612, -4.59912e-05]

depth_scale: 0.0010000000475.

I have a depth and color images aquired by the camera, and I want to allign the depth image to the color image. The code Im using is:

depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_scale)
color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_point)
color_pixel = rs.rs2_project_point_to_pixel(color_intrin, color_point)

The code obtains the point cloud of the scene with the depth camera coordinates ("rs.rs2_deproject_pixel_to_point"). Then I change from the depth camera coordinates to the color camera coordinates (with the "rs.rs2_transform_point_to_point"). Finally, I project the last poincloud using the color camera intrinsic values. At the end I should have a 424x240 depth image alligned to the color image but instead I have a 521x292 image.

RealSense-Customer-Engineering, 2018-10-03:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

If you like to re-project depth to color pixel, need to modify as follows.
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth value * depth_scale)

Did you also try align sample?

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

Did you try "align.m" in matlab to see how it works and fulfill your requirement?

alt01, 2018-11-10:
I dont know how to use the function specified on align.m, it seems to be a method from a class.
 I already have the color and depth streams on matlab, but im having troubles with the align function.

[image]

RealSense-Customer-Engineering, 2018-11-19:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

See #2710 to fix the error in align method. Then, you can try the process function like below code to align depth to color.

    align_to = realsense.stream.color;
    alignedFs = realsense.align(align_to);
    .....
     fs = pipe.wait_for_frames();
     aligned_frames = alignedFs.process(fs);

alt01, 2018-12-11:
Is it possible to align a depth and color images that i have stored on my computer?
This is because I took some samples (depth and color images) with a D435 depth camera, and I need to work with this samples.
I have the intrinsic and exrinsic values of the camera.

MartyG-RealSense, 2018-12-11:
@alt01  This discussion linked to below may be useful to you:

https://github.com/IntelRealSense/librealsense/issues/2534

apachebalou, 2018-12-14:
> [Realsense Customer Engineering Team Comment]
> Hi @alt01,
>
> See #2710 to fix the error in align method. Then, you can try the process function like below code to align depth to color.
>
> ```
> align_to = [...]

This is exactly what i'm looking for, however I get the following error message when executing the lines above:
"Error using realsense.align/process (line 17)
Expected input number 2, frame, to be a scalar.

Error in realsense.depth_example_stream_testingalign (line 41)
    aligned_frames = alignedFs.process(fs);
"
Just like alt01 I'm on D435 and have the depth and color streams running. Any idea what can be the issue? I was expecting align/process to be happy with a frameset...

sebawski, 2019-01-03:
> This is exactly what i'm looking for, however I get the following error message when executing the lines above:
> "Error using realsense.align/process (line 17)
> Expected input number 2, frame, to be a [...]

Hello apachebalou!

Did you solve the problem?
I have the same problem

apachebalou, 2019-01-03:
> Hello apachebalou!
>
> Did you solve the problem?
> I have the same problem

Hi sebawski,

Actually I just commented away validateattributes check in align.m and it worked. I dont understand the root cause, but seems to work at least.

sebawski, 2019-01-03:
> Hi sebawski,
>
> Actually I just commented away validateattributes check in align.m and it worked. I dont understand the root cause, but seems to work at least.

Thank you! finally the code worked

wkunwkun, 2019-01-17:
@apachebalou Thank you! It worked

closed 2019-02-19 by RealSenseCustomerSupport

JimmyFroost, 2019-04-29:
Hello,

I am trying to understand and apply your code but I'm really struggling with this.
I manage to have both rgb and depth streaming, now I 'd like to align RGB to this depth data, but I don't know how to process your discussion.

How do you display the expected result : a colorized pointcloud ?
Can you please share a bit more of your code ?

redyoung75, 2021-07-29:
Hi 

Does anyone can help me to use built-in alignment feature between. color and depth image in VB.NET? I have seen many examples in python and C but I really haven't seen any VB.NET sample code for image alignment... Any help will be very appreciated!
Thanks.

adambiomech, 2023-10-16:
I am getting the error: 
_Error using realsense.align
Expected input to be a scalar._

I am passing realsense.align a vector uint8 variable which is the data from the color frame set. I can only pass a number or a realsense.stream into the class so cf is not an option due to it being a video_frame variable. I cannot find any examples of using the align class when not using the stream function. 

```
 % Obtain frames from a streaming device
        fs = pipe.wait_for_frames();

        % Produce pointcloud
        if (depth.logical() && cf.logical())
            
        % Align depth frame to Color
            cf = fs.get_color_frame();
            dd = cf.get_data();

            align = realsense.align(dd);
        end
```