realsenseai/librealsense #2456: librealsense on Android + D415: Command transfer over XU?

xtrawurst, 2018-09-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Android 8.0 | 
| Language                            |  C/C++    | 
| Segment			|  Phones & Tablets | 

### Issue Description
We're having some trouble running a D415 on Android.
Some debugging revealed that in `ds5_device::init()` in ds5-device.cpp lines 322-335 the second path is taken due to `group.usb_devices` being empty.

That in turn is because `try_fetch_usb_device()` in ds5-private.cpp wants 6 interface points for the D415 (line 190) (`result.mi` must be 5 and `result.mi` is always `bNumInterfaces - 1`).

I have checked on both Android and Windows and the # of interface points for the D415 is always 5.

My question is: Is going through XU a problem or would it be much better taking the first path in `ds5_device::init()`? Also in which circumstances would the D415 have six interfaces?

dorodnic, 2018-09-28:
Hi @xtrawurst 
Cameras provided under NDA would have the extra interface. From librealsense point of view it and the XU are functionally equivalent.

xtrawurst, 2018-09-29:
@dorodnic Thanks. Are there any downsides w.r.t. USB bandwidth / latency when using XU?

dorodnic, 2018-09-29:
Well, if you need to transfer large messages, like reading / writing flash memory blocks, the XU method can be noticeably slower. However, this is exactly the type of things you no longer need to do with the final product. On the other hand, there are downsides to leaving the extra interface - on Windows the inbox WinUsb driver did not support [D3-cold](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/device-sleeping-states) power-state making the whole thing a bit less efficient. On Linux, the extra interface is the one forcing you to install udev-rules and mess with the permissions system. 

The only grey area are the advanced-mode commands. When switching presets, a lot of register-sets are being sent to the firmware. I'd say firmware can reconfigure everything in 60-300ms, rest of the time is spent in transfer. We are working on adding a "single-command" preset control that would do most of that work at the firmware level. Depending on the preset, this can speed things up a bit. (I have a [work-in-progress commit](https://github.com/dorodnic/librealsense/commit/013ad5a3bb553b1547f944847daa1696087dd629) that shows how to access this firmware control, but it is not yet **officially** supported)

xtrawurst, 2018-09-29:
@dorodnic Thanks much for the detailed answer. From the code it looked a bit like XU is the "fallback", but now I understand it's actually the default for release products.

dorodnic, 2018-10-08:
Hopefully this is resolved for you. Feel free to raise more questions about the implementation

closed 2018-10-08 by dorodnic