realsenseai/librealsense #2457: Mapping infrared images to color images in the libRealSense2 lib

starspder, 2018-09-29:
|---------------------------------|------------------------------------------- |
| Camera Model | { Realsense D435 depth camera} |
| Operating System & Version | {Win 10} |
| Platform | PC |
| Language | {c++ } |

 I want to align with the left-infrared camera and the color camera.

the align function provided by the libRealSense2 library has only the ability to align depth and color.

and I heard that..
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
Customer Engineering Team Comment] @Panepo The align class used in librealsense demos maps between depth and some other stream and vice versa. We do not offer other forms of stream alignments.

But one suggestion for you to have a try, Basically the mapping is a triangulation technique where we go through the intersection point of a pixel in 3D space to find its origin in another frame, this method work properly when the source data is depth (Z16 format). One possible way to map between two none-depth stream is to play three streams (Depth+IR+RGB), then calculate the UV map for Depth to Color, and then use this UV map to remap IR frame ( remember that Depth and left IR are aligned by design).
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
it is aligned with left-infrared and initiated depth image, so if you find only UV map from depth to color, you can map it.

I can use Align Function but How do I get a depth to color UV map when I align?

Is there a function in librealsense2 that can return a UV map?

I am sorry that English is poor because I'm Korean. but I need your precious answer.

MartyG-RealSense, 2018-09-29:
Your English is very good, don't worry.  :)

I hope you can find something useful in the discussion below.  Please read downward through the comments from the point in the discussion that is linked to.

https://communities.intel.com/message/530761#530761

starspder, 2018-09-29:
thank you for the reply. I see the contents of the link.

This is what I understand.

ㅡ
lternatively, I believe you can first align the depth image to colour using Align, then pass that aligned depth image to the point cloud to get your point cloud aligned to the colour image if you'd prefer. Then getting the colour of a vertex or the vertex of a colour pixel is simple as they will have the same indices of their respective arrays. (so (x,y) in colour image has corresponding vertex points[x + y * colourWidth] and vice-versa)
ㅡ

ㅡ
I believe the point cloud is aligned to the depth image, which itself is aligned to one of the IR images (from streaming with the viewer app and putting my finger over the lenses in turn, it appears to be the left lens as you're looking at it, which is called "Infrared 1" in the UI).

From what I understand, the standard point cloud is equivalent to MapDepthFrameToCameraSpace, using align will be like MapDepthFrameToColorSpace, and passing the aligned depth image to pointcloud should be like MapColorFrameToCameraSpace.

We're fast approaching the limits of my knowledge here so hopefully someone else will be able to double-check my assertions!
ㅡ

In this passage I became convinced that the depth image was aligned to the left-infrared image.

so 

1. Align the depth image to the color image by using Align function.

2.   rs2::pointcloud pc;
     pc.map_to(m_oFrameset.get_depth_frame()); //all points have coordinate system of the depth camera.

-> the depth image is already Aligned to color image.

 3.  m_oInfraredPointsInDepthCoords = pc.calculate(left_Infrared_frame); //calculate the coordinates of the Infrared frame in the depth camera coordinate system

     pVerticesTmp = m_oInfraredPointsInDepthCoords.get_vertices();

     pVertices = (void *)pVerticesTmp;

To summarize, we first map the depth image to the color image, map the aligned depth image to the pointcloud, which Stores an expression that aligns the initial depth image to the color image and

I can change the infrared image to a aligend depth image. Since the depth image is already aligned to the color image, it can be said to be in color image and aligned state too.

Is it right that I think? If I have any misunderstanding, please give a keen point.

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Ticket being closed due to inactivity for 30+ days

closed 2018-12-12 by RealSense-Customer-Engineering

shinokain, 2019-12-20:
Hello @starspder, sorry for reopening this but i encountered the same problem, have you managed to do it?