realsenseai/librealsense #2458: How to deproject a pixel in color frame to 3d coordinates using D400 series?

Superlee506, 2018-09-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Linux (Ubuntu 16）  |                               | 
| SDK Version                            | 2.10.1| 
| Language                            | Python | 

### Issue Description
As refers to the document of [Projection in RealSense SDK 2.0](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0), I know the we can't directly deproject a pixel from RGB to 3d coordinate. I wonder whether the follow codes are right to deproject from RGB to 3d coordinates.

```
##Firstly align the depth frame to color frame
align_to = rs.stream.color
align = rs.align(align_to)
while true:
    # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
         frames = align.process(frames)
         depth_frame = frames.get_depth_frame()
         depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
         color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
         color_frame = frames.get_color_frame()
          if not depth_frame or not color_frame:
              continue
           #Get the  depth value
           depth = depth_frame.get_distance(100, 100)
           depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, [100, 100], depth)
```
I refer to [1904](https://github.com/IntelRealSense/librealsense/issues/1904), but it ignores the process with alignment. But I'm not sure with above codes, because after the alignment, can we still use the "depth_intrin" to depreject a depth pixel?

ev-mp, 2018-10-03:
That is correct.
Note that after align.process(...) the `depth` frame transform in terms of intrinsic and possibly resolution.
What you'd expect is that in the above code `depth_intrin == color_intrin`.
After that, de-projecting the modified depth using the modified intrinsic will work correctly (with small fluctuations)

Superlee506, 2018-10-16:
@ev-mp Thanks~ Copy that.

closed 2018-10-16 by Superlee506