realsenseai/librealsense #2459: The erros when I run the example of unity(Unity package file (Intel.RealSense.unitypackage))

coldcoda, 2018-09-30:
OS:Win10
Unity 2017.3.1
RealSense2.16.1

When I run the project and click the button in the scene called"StartHere",here is a error shows"Unknown Shader channel count".
[image]

So, why?I need help,thank you!

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Please try to upgrade your Unity3D editor. I do not find this issue in Unity 2018.2.13.
cc @matkatz

coldcoda, 2018-10-23:
@RealSense-Customer-Engineering @matkatz 
Yes,after I updated my Unity Editor to 2018.2.13,there is no such issue.
But,when I run this scene,It has the error like this:
The Scene:
[image]
Error:
[image]
(Is this related to the USB interface?I have used USB3.0)
And,When I run this scene,my Unity Editor will crash and stop working:
The Scene:
[image]
crash and stop working:
[image]

RealSense-Customer-Engineering, 2018-11-13:
[Realsense Customer Engineering Team Comment]
Is there any update?

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
Yes, please have a try.

closed 2018-11-27 by RealSense-Customer-Engineering