realsenseai/librealsense #2460: No depth info from D435

mitchbte, 2018-09-30:
I'm trying to use the D435 to get depth information. I'm running arch linux natively and ubuntu 18 inside a docker container. When I first got the camera, I plugged it in, installed the apt packages and ran the realsense viewer - everything worked no issues.

I then updated the firmware to version 5.9.13, pulled the librealsense master and tried to build and run the examples. The rgb camera worked but not the depth camera. I also tried running the realsense viewer again and it will only let me display the infrared camera feed, there is no depth option.

I've updated to the latest 5.10.3 firmware. I've also tried building and install the drivers from librealsense master (e46e09afc0c6f888e0042260e635e10062239381). Still no depth feed, only rgb and infrared.

Currently running kernel: 4.18.9-arch1-1-ARCH

I'm not sure how to start debugging this issue it seems like a firmware issue, I would be happy to downgrade but I don't know which version comes pre-installed on the device.

Running the rs-align example results in the following error:
```
root@2dfa37166c38:/realsense/camera/realsense/librealsense/examples# ./build/align/rs-align 
No Depth stream available
```

Here is my device info (the video0 and video1 errors are because those devices are my laptop camera and don't exist in the docker container):
```
root@60e6f5bac9c9:/usr/local/bin# rs-enumerate-devices 
 30/09 11:52:34,426 ERROR [139948856820672] (types.h:179) Cannot identify '/dev/video1 Last Error: No such file or directory
 30/09 11:52:34,599 ERROR [139948856820672] (types.h:179) Cannot identify '/dev/video0 Last Error: No such file or directory
 Device info: 
    Name                          :     Intel RealSense D435
    Serial Number                 :     819312070516
    Firmware Version              :     05.10.03.00
    Recommended Firmware Version  :     05.10.03.00
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2:1.0/video4linux/video2
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B07
    Usb Type Descriptor           :     3.2

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2    1280x800      @ 30Hz  Y8
    Infrared 1    1280x800      @ 30Hz  Y8
    Infrared 2    1280x800      @ 25Hz  Y16
    Infrared 1    1280x800      @ 25Hz  Y16
...

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color         1920x1080     @ 30Hz  RGB8
    Color         1920x1080     @ 30Hz  Y16
 ...
```

RGB works
```
root@2101b58c22c3:/usr/local/bin# rs-color 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 819312070516
    Firmware version: 05.10.03.00

RGB frame arrived.
First 10 bytes: 10 12 00 10 12 00 11 13 01 0e 
Frame No: 1
Timestamp: 61182.274000
Timestamp domain: Hardware Clock
Time of arrival: 1538311002511

RGB frame arrived.
First 10 bytes: 0d 10 00 0f 12 00 10 12 00 10 
Frame No: 2
Timestamp: 61215.873000
Timestamp domain: Hardware Clock
Time of arrival: 1538311002544
```

Depth hangs
```
root@2101b58c22c3:/usr/local/bin# rs-depth 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense D435
    Serial number: 819312070516
    Firmware version: 05.10.03.00
<-- Hangs here
```

gauravbansal98, 2019-01-12:
@mitchbte  can you please tell me how you installed realsense inside docker. While following the official tutorial for installing sdk for ubuntu I am facing some errors. Thanks

mitchbte, 2019-01-12:
You should make a new issue, rather than commenting in a closed one. What are the issues that you're having? I think I used 2 different methods, here is my Dockerfile. I'm currently using ros to use the realsense ros bag logger. If i remember correctly you can just use the commented out section at the bottom the ubuntu18 image

```
FROM ros:melodic-ros-base-bionic

#install ALL Python packages with pip

RUN apt update; \
    apt upgrade; \
    apt install -y apt-utils build-essential cmake unzip pkg-config git vim tree \
                   python3.7-dev python3-pip; \
    ln -sf /usr/bin/python3.7 /usr/bin/python; \
    python -m pip install --upgrade pip; \
    python -m pip install --upgrade setuptools wheel numpy;

# RUN apt install -y libopencv-dev;
RUN apt-get install -y libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev; \
    libavcodec-dev libavformat-dev libswscale-dev libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libdc1394-22-dev; \
    x264 libx264-dev;

RUN git clone -b '3.4.3' --single-branch https://github.com/opencv/opencv.git /opencv; \
    git clone -b '3.4.3' --single-branch https://github.com/opencv/opencv_contrib.git /opencv_contrib; \
    cd /opencv; \
    mkdir build; \
    cd build; \
    cmake -DCMAKE_BUILD_TYPE=Release -DOPENCV_EXTRA_MODULES_PATH=/opencv_contrib/modules .. ; \
    make -j 4 ; \
    make install; \
    cd /; \
    rm -rf /opencv /opencv_contrib;

COPY librealsense /librealsense

RUN cd /librealsense; \
    rm -rf build; \
    mkdir build && cd build; \
    cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true; \
    make -j 4; \
    make install; \
    cd /; \
    rm -rf /librealsense;

RUN apt install -y ros-melodic-rqt-bag; \
    pip install rospkg pyyaml PySide2 pycryptodome gnupg defusedxml netifaces lz4;

#RUN apt install -y software-properties-common; \
#    apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE; \
#    add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u; \
#    rm -f /etc/apt/sources.list.d/realsense-public.list; \
#    apt-get update; \
#    apt-get install -y librealsense2-dkms librealsense2-utils librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-dbg;

```