realsenseai/librealsense #2462: Question : how to read saved binary frames ?

qrasmont, 2018-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Ubuntu 18.04.1 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-33                                         | 
| Platform                                 | intel NUC  |
| SDK Version                            | 2.0 (Build 2.16.1)                       | 
| Language                            |  C++                         | 
| Segment			|  other                          | 

### Issue Description

Hi,
I can succesfully write frames as binary using this methods provided in another issue :

```
bool save_frame_raw_data(const std::string& filename, rs2::frame frame)
{
    bool ret = false;
    auto image = frame.as<rs2::video_frame>();
    if (image)
    {
        std::ofstream outfile(filename.data(), std::ofstream::binary);
        outfile.write(static_cast<const char*>(image.get_data()), image.get_height()*image.get_stride_in_bytes());

        outfile.close();
        ret = true;
    }

    return ret;
}

```

How can I recreate frames from those saved binaries ?

Thanks!

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
Hi @qrasmont,

Please review #2231 to see if it fulfill your request to process the RAW data.

qrasmont, 2018-10-02:
Hi,

#2231 is not quite what i am trying to achieve.
Actually, the feature request in #2384 would do exactly what I need. 

Basically, what I want to do is frame post processing (not with a frame queue, in a separate program).

I want is to get from the binary files (or any other format that I could save to) an rs2::frame of depth and color that would allow me to do step like 
```
rs2::pointcloud pc;
pc.map_to(color);
auto points = pc.calculate(depth);
...
```
color and depth being rs2::frame objects.

A simple `static_cast` of the binary file as a frame object does seem to cut it.

ogoshen, 2018-11-18:
@qrasmont notice that an `rs2::video_frame` is more than just a container for pixel data,  you can't just cast binary data to a frame. See the **rs-software-device** sample for a way to generate frames from external data. 

Having said that, you still won't be able to process those frames and get a pointcloud because of missing data like sensor intrinsics & extrinsics. 
Either save and load those yourself or consider using the rosbag format as container instead of raw files, 
as it also serializes sensor, stream, and frame information (intrinsics, extrinsics, timestamps...) it also supports compression and is more interchangeable.

So for your use-case of offline processing, I suggest you try the following:

1.  Record a live stream, either from the viewer, or see the **rs-record-playback** sample
2.  Write a program that creates 2 devices: 
*a*.  `rs2::playback` to play back the recording (with `set_real_time(false)`)
*b*. `rs2::software_device` wrapped in a `rs2::recorder` to write processed frames back to disk
3. Configure the software device (*b*) according to the playback device (*a*), see **rs-software-device** sample for details
4. Grab frames from playback device (*a*), process them and send to the software device (*b*)

If your `rs2::software_device` was properly configured, playing back the resulting (processed) rosbag should work just like a recording of a hardware device, including processing blocks like pointcloud and alignment (but then again, you could just do those offline with the rest of the processing... 😄 )

@dorodnic maybe this functionality could be simplified with some kind of pass-through device and custom post-processing blocks.

qrasmont, 2018-12-03:
I'll investigate that thanks !

shahabe, 2018-12-31:
@qrasmont, could you manage to get your results?
I also need to save the whole frame with all its meta-data in order to read it again in my another program and perform my tasks.
Thank you.

qrasmont, 2019-10-22:
@shahabe  @josephdanielchang 
Sorry guys for not responding, I haven't been active on our realsense project for a while.
I'm afraid I will not help you. Our goal was to reduce the processing as much as possible when capturing (as we're in a multicam setup) but we managed to optimize in other ways so I never investigated this further.

ev-mp, 2019-11-04:
@qrasmont , thank you for the update.
We'll close the ticket accordingly.

closed 2019-11-04 by ev-mp