realsenseai/librealsense #2463: 3D reconstruction of any random object, D435, C++

EmmanuelCos, 2018-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  Windows 10  | 
| Platform                                 | PC  |
| Language                            |  C++ with opencv }                          | 

Hello,
I would like to do 3D reconstruction of any random object. The goal is to take several pictures around the object, and then reconstruct it, as does Cappasity https://cappasity.com/.
Do we have any source code for that kind of thing?
Thank you,
Emmanuel

MartyG-RealSense, 2018-10-01:
A pre-made application used with the 400 Series cameras that works like Cappacity is LIPScan.

https://www.youtube.com/watch?v=N14Pi6z-MkE

EmmanuelCos, 2018-10-01:
I already heard of this one but the sofware isn't free and I don't think that you can have access to the source code.

EmmanuelCos, 2018-10-01:
Hello Marty,
Yes I have seen this discussion, and I understand the theory on how I should do that.
For me the difficult step would be to merge all the different pointclouds taken with different angles, and I didn't find any opensource code for that, even in the provided links.
Best regards,
Emmanuel

MartyG-RealSense, 2018-10-01:
The merging of point clouds into a single cloud was mentioned in Intel's recent RealSense multiple camera webinar.  They said: "[the software] Vicalib can do this, but there is a simpler approach, which will work in 90% of cases.  This is is to take the point cloud from every one of the cameras and then do an Affine Transform.  Basically, just rotate and move the point clouds, in 3D space, and then once you've done that, you append the point clouds together and just have one large point cloud".

Regarding aligning clouds with Vicalib, the webinar said "it uses a board that you show to each of the cameras in turn., and it establishes overlapping regions to then minimize the pose of each of those together".

If you want to check out the mentioned Vicalib, please see this link:

https://github.com/arpg/vicalib

EmmanuelCos, 2018-10-01:
OK thank you, where can I find the page of Intel's recent RealSense multiple camera webinar please?

EmmanuelCos, 2018-10-01:
Thank you Marty for your answers, but Vicalib seems to be only available on UNIX and Mac. 
Does anyone know any open source software available on windows?

xtrawurst, 2018-10-04:
Check out this page:
https://realsense.intel.com/software-for-intel-realsense/

EmmanuelCos, 2018-10-04:
Hello xtrawurst,
What I was looking for was a free open source code with ideally an easy to use software implementing the functions to see if the reconstruction fits my expectation. (none of the softwares on this page are free)
I'm not sure this exists, but anyway, the library Open3D that MartyG advised me has some good reconstruction functions, that will probably work for what I need. PCL also has some, but I didn't tested them.
Regards,
Emmanuel

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@EmmanueCos How about your test with the library Open3D? Any other questions about this ticket? Please update. Thanks!

EmmanuelCos, 2018-10-19:
Hello,
Open3D has very good reconstruction functions and a good tutorial too. You can have a look at this page: 
http://www.open3d.org/docs/tutorial/ReconstructionSystem/system_overview.html
Most of the functions are developped in C++ and Python too. But the main tools like Reconstruction system are only in Python. So I had to transcribe them in C++, but it works really well for what I need to do.

Anyway, it would be great that Intel develops an easy to use reconstruction tool for their realSense cameras, as it is something really expected, when we buy a 3D camera, and in this context, if you are not a developper and don't have any knowledge in mathematics and geometry, can be almost impossible to achieve!

Thank you for your help,

Emmanuel

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
@EmmanuelCos Thanks for your valuable suggestions! Will try to evaluate.

HippoEug, 2018-10-26:
Cannot wait for the integration of Open3D in future releases as seen [here](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.2)!

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
@EmmanuelCos Is it ok for you to close the ticket at this moment?Thanks!

EmmanuelCos, 2018-11-16:
Ok, thank you for your help!

closed 2018-11-16 by EmmanuelCos

legel, 2018-11-16:
Just a quick check on the status of the Open3D integration? Is there an alpha push somewhere?

MartyG-RealSense, 2018-11-16:
The most recent statement I know about it was from Dorodnic on October 2.  "We are planning to improve the integration with it in one of the upcoming releases, but we continue to work based on community feedback and offer incremental improvements at regular time intervals".

I cannot see any Open3D references in the list of commits for the upcoming SDK version, so it probably is not coming in the month of November.

metanamo, 2018-12-21:
Hello EmmanuelCos,

have you found a way to merge the clouds?
I also try to get three D415 aligned. It should be the same code to find the camera's pose.
This is still giving me a hard time. Any open source I could find did not the trick (on Windows). 

Using opencv's calibrateCamera with a checkerboard calibration pattern gives you the 
OutputArrayOfArrays rvecs, OutputArrayOfArrays tvecs
of each camera - the rotation and translation vector for the camera pose.

However I can't figure out how to use the rvec and tvec to get the pointclouds merged.

Good luck...

EmmanuelCos, 2019-01-02:
Hello Metanamo,

I haven't merge point clouds taken by different cameras, but several images taken with the same camera at different views, without knowing a priori the rotation/translation of the camera at each different views.
To register the different point clouds, I used Open3D reconstruction method. 
http://www.open3d.org/docs/tutorial/ReconstructionSystem/system_overview.html
It worked pretty well if the different views were taken pretty close from one another.

I will soon have to merge four pointclouds taken by four distinct cameras, knowing a priori the distance and angles between each camera. What I am planning to do is to rotate and translate the point clouds so that they are in the same coordinate system, and then apply the same reconstruction method. But I didn't test it yet and am not sure it is the best way of doing this.

You could maybe try to contact directly intel support as they have already developed multicam synchronisation: 
https://realsense.intel.com/intel-realsense-volumetric-capture/
I don't know if they would share there code.

Best regards and good luck,

Emmanuel

mikeNike9, 2019-01-17:
I know this is closed but @EmmanuelCos is correct on this. This is a simple way of combining the clouds into a mesh by calibrating the system to a central origin. You can do this by determining the translation vectors for each POV prior to each scan and basically just DO MATH haha. 
If you want to get crazy you can try and do this with one camera by moving the object several degrees at a time. Capture each frame and append to final points array. The only thing you will run into is this technique will not get rid of flying pixels (noise) so you will need to add a few flying pixel algos to help smooth the image. An easy way to minimize this is to do an extreme pass through filter to only let the front face of the object to be appended to final list. 

Hope this helps at all. 
FILTER FILTER FILTER

EmmanuelCos, 2019-01-17:
Hello @mikeNike9 , thank you for your comment.

In this method, the only thing that's frightening me is how the generated point cloud will look like if there is a little error measure, that is to say if we don't have exactly the right angles and distances between the different cameras. 

Do you know what would be the best way to prevent that? Should we perform a registration algorithm after we set all the different point clouds in the same coordinate system?

Best regards,

Emmanuel

mikeNike9, 2019-01-18:
@EmmanuelCos  

I would agree with you here. For me I use this type of method to calculate a smallest possible cube the object can fit into.
I havent found the need for a registration algo because my angles are very precise when I tested this theory and my tolerance at the end only needs to be + - 1 cm. I dont usually go that far and havent ventured down the path of writing my own registration algorithm. 
If you are thinking of writing an ALGO to do this please share and I can test with you. 
Again its all about getting rid of the bad noise pixels and trusting your point-cloud from there. 

**CALIBRATION**
If you are using multiple cameras I can help you calibrate better to find the correct distance and angles of the cameras.
You should use a small card board box with only flat sides and crisp edges. 
You will have a none measurement the volume of the object should be. 
after you have run your scans and meshed the clouds together you can run a simple 3D convex hull algo to see if your measurement of the box comes out correct. This practice allows you to see if you are overlapping or not calibrated correctly. Once you have your system calibrated to your liking you should be good to go. 

May I ask what your end goal is with this system?

EmmanuelCos, 2019-01-18:
My goal is to have a 3D representation of small objects (~10cm length, ~4cm width, ~4cm height).

EmmanuelCos, 2019-01-24:
@mikeNike9 
Hello Mike, 
I've done my first tests and it appears that the theoretical distances I have between each camera has an error measure that leads to a gap between each image when I try to reconstruct a point cloud only by doing rotations and translations. 
I am using a registration algorithm, coming from open3D, that I can provide you if you need, that slightly improves the result, be it is not perfect, though. 
So I will have to do the calibration as you described in your previous message. Do you have any sample code I could use to do the calibration between each camera?
In parallel, I will ask for the same question in a separated issue, as it is an important subject in itself, and this one is supposed to be closed.
Thank you and regards,
Emmanuel

mikeNike9, 2019-01-24:
@EmmanuelCos
If you could share your registration Algo that would be awesome. I have a
home project that I would love to try this on.
However, I cannot share my calibration code because it is proprietary
software owned by the company I work for.
Can you share with me how you have tried to calibrate so far. Maybe I can
help you.
Could be distance to your central origin is incorrect for each sensor. In
my experience the calibration distance to your global origin is slightly
smaller than the true distance.
Also you will need to take into consideration where you should be measuring
from on the sensor. The middle  of the sensor to the origin. These
variables could be causing a small error.

Can you explain more on what you mean by gaps in the point clouds. Are they
too close or too far apart?
An image of it all would be easy for me to help.

Mike

-- 
Best Regards,

*Mike Eichenberg Jr.*
Research & Development Engineer

P: 913.967.5073 | M: 913.220.8619 | FreightSnap.com
7907 Bond Street | Lenexa, KS 66214

CONFIDENTIALITY NOTICE: This message, including any attachments, originates
from, and contains confidential and proprietary information of FreightSnap,
LLC. It is intended solely for the use of the individual(s) named above. If
you believe you received this message in error or if the content of the
email does not relate to prior conversations/expectations please delete it
(and any attachments) from your systems and notify the sender promptly via
email or direct contact at our main office number 1- <(913)%20213-5685>
913-967-5073 <(913)%20967-5073>. Any unauthorized review, use, disclosure,
copying or distribution is prohibited. FreightSnap® is a registered
trademark