realsenseai/librealsense #2469: Getting Depth Data in CSV Format using C++?

HippoEug, 2018-10-02:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

## Issue/Question
Hello again,

I would like to get the coordinates (x, y, z) values in csv format. Especially the z depth data in order to get an average of the depth on points/coordinates that I am interested in. 

[image]

I have captured the picture using the SDK example, rs-save-to-disk.cpp. I also wrote a simple application to let user select points of the picture and returns the (x, y) coordinates. I am trying to get the z values of these (x, y) coordinates..

1. I understand there is an example, rs-depth which can return the exact coordinates of these points I am interested in, but however this function only works in real time. Hence, I can't select the coordinates I am interested in and look up the z depth value afterwards (after data capture).

NOTE: I was thinking of a CSV table with all 921,600 coordinates (x, y, z), example:
[image]

2.  As I have little programming experience, I thought of a idea which does not work practically. Basically, the depth image has a resolution of 1280x720 pixels, or 921,600 pixels in total. But it is not practical to write a loop that loops 921,600 times just to acquire the depth data.. way too slow. Any idea how can I overcome this? 

3. I have taken a look at the rs-pointcloud example, but I do not have enough understanding to convert any of those depth data into a readable CSV format. 

4. A suggestion I have read is to record it in a bag file and convert it using the SDK rs-convert tool into a CSV format. However, I am having trouble building that.. so have not tried it. 

My goal is to measure the volume of an indentation, but firstly I would like to know the depth of the surrounding surface around the indentation in order to offset the depth.

Appreciate any feedback or guides, please make it easier to read as I am a complete noob haha! Thank you in advance!!

ev-mp, 2018-10-03:
Hello @HippoEug ,
If you're interested in offline analysis then rosbag records and `rs-convert` are the tools of choice .
 
Rosbag records can handle all the related streams and the converter supports several formats oriented for both 'excel' and 'matlab'-friendly users.
Moreover, rosbags are not only convenient but actually necessary if you're going to cross-correlate (sync) streams, e.g. look for features in color frame, and then analyse the corresponding area in the depth data.

RealSense-Customer-Engineering, 2018-10-25:
[Realsense Customer Engineering Team Comment]
Hi HippoEug,

Your current question is addressed in https://github.com/IntelRealSense/librealsense/issues/2584.
Can we are closing this one for now?

Thanks!

HippoEug, 2018-10-26:
Yes, please. thank you. Close #2469

closed 2018-10-26 by RealSense-Customer-Engineering