realsenseai/librealsense #2471: Using librealsense library in own projects

Lasy92, 2018-10-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |  MacOS  | 
| Platform                                 | Macbook Pro  |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C++ (Xcode project)                         | 
| Segment			|  Others                          | 

### Issue Description
I have successfully compiled and ran the provided examples following the guide shown here: [https://github.com/IntelRealSense/librealsense/tree/legacy].

Now I want to add the librealsense library to my own project, where I want to access all the functionalities provided with the librealsense library and use them together with a QT GUI and OpenCV functions. 

I copied the library folder to the xcode working directory, and added the library search path to the library location. Afterwards I tried to `#include <librealsense2/rs.hpp>`, and the compiler was able to locate the library, but then several errors emerged shown below:
[image]

What configuration do i need to do in order to compiler my project with the librealsense library?

Thanks in advance, Lasse

Lasy92, 2018-10-03:
I got around this issue by moving `#include <librealsense2/rs.hpp>` up to be the first include, since it seems to have conflicting names with QT, [https://github.com/IntelRealSense/librealsense/issues/1059](url), but now a new problem ermerged.

The program gets stuck in a thread, and writes:

> dyld: Library not loaded: @rpath/librealsense2.2.dylib
>   Referenced from: /Users/lassechristensen/Desktop/Siemens/RealsenseTest/build/Debug/RealsenseTest
>   Reason: image not found

I would really appreciate if I could get a simple guide of how to setup the Xcode build settings, and thereby use the library. 

Thanks, Lasse

HippoEug, 2018-10-04:
I experienced the same issues with you but on the Windows Platform. It is virtually not documented on how to go ahead about adding those library files.. where the only suggestion I keep seeing is just to use CMake!!

Anyways, the link [https://github.com/IntelRealSense/librealsense/tree/legacy] you provided I believe is the SDK Version 1, whereas #include <librealsense2/rs.hpp> is for the SDK Version 2. You should use #include <librealsense/rs.hpp> if you're using the legacy version (V1).

## EXTRA INFORMATION:
[Here](https://github.com/HippoEug/Intel-Realsense-D435) is how I did it for Windows, but I believe it will only work with Visual Studio, not XCode unfortunately. After doing these steps I could use the <librealsense/rs.hpp>, but still unable to use OpenCV etc.

[Then](https://github.com/HippoEug/OpenCV) I had to figure out how to add OpenCV library to the Realsense library just to get some things to work.. but again I believe it is exclusively for Visual Studio.

Lasy92, 2018-10-04:
Thanks for the information, unfortunately i don't think it correspond to the issues experience on the MacOS system. The guide which you provided is really nice, but can't be used to solve the problem on MacOS. 

I am currently using qMake to create the xCode project and is set up as: 
`#-------------------------------------------------
#
# Project created by QtCreator 2018-10-03T14:38:42
#
#-------------------------------------------------

QT       += core gui

greaterThan(QT_MAJOR_VERSION, 4): QT += widgets

TARGET = camera_software
TEMPLATE = app

# The following define makes your compiler emit warnings if you use
# any feature of Qt which has been marked as deprecated (the exact warnings
# depend on your compiler). Please consult the documentation of the
# deprecated API in order to know how to port your code away from it.
DEFINES += QT_DEPRECATED_WARNINGS

# You can also make your code fail to compile if you use deprecated APIs.
# In order to do so, uncomment the following line.
# You can also select to disable deprecated APIs only up to a certain version of Qt.
#DEFINES += QT_DISABLE_DEPRECATED_BEFORE=0x060000    # disables all the APIs deprecated before Qt 6.0.0

CONFIG += c++11 

QMAKE_INFO_PLIST = my_Info.plist

SOURCES += \
        main.cpp \
        mainwindow.cpp \
    	videostreamthread.cpp 

HEADERS += \
        mainwindow.h \
    	videostreamthread.h \
    	Libraries.h \
    	librealsense2/rsutil.h \
    	librealsense2/rs.hpp \
    	librealsense2/rs.h \
    	librealsense2/rs_advanced_mode.hpp \
    	librealsense2/rs_advanced_mode.h \
    	librealsense2/hpp/rs_types.hpp \
    	librealsense2/hpp/rs_sensor.hpp \
    	librealsense2/hpp/rs_record_playback.hpp \
    	librealsense2/hpp/rs_processing.hpp \
    	librealsense2/hpp/rs_pipeline.hpp \
    	librealsense2/hpp/rs_internal.hpp \
    	librealsense2/hpp/rs_frame.hpp \
    	librealsense2/hpp/rs_device.hpp \
    	librealsense2/hpp/rs_context.hpp \
    	librealsense2/h/rs_types.h \
    	librealsense2/h/rs_sensor.h \
    	librealsense2/h/rs_record_playback.h \
    	librealsense2/h/rs_processing.h \
    	librealsense2/h/rs_pipeline.h \
    	librealsense2/h/rs_option.h \
    	librealsense2/h/rs_internal.h \
    	librealsense2/h/rs_frame.h \
    	librealsense2/h/rs_device.h \
    	librealsense2/h/rs_context.h \
    	librealsense2/h/rs_advanced_mode_command.h

FORMS += \
        mainwindow.ui

INCLUDEPATH += \
				/usr/local/include
				
LIBS += \
		/usr/local/lib/ 
		/usr/local/lib64

QMAKE_LFLAGS += \
				-lopencv_calib3d -lopencv_core -lopencv_features2d -lopencv_flann \
				-lopencv_highgui -lopencv_imgcodecs -lopencv_imgproc \
  				-lopencv_ml -lopencv_objdetect -lopencv_photo -lopencv_shape \
				-lopencv_stitching -lopencv_superres -lopencv_video \
				-lopencv_videoio -lopencv_videostab
				

# Default rules for deployment.
qnx: target.path = /tmp/$${TARGET}/bin
else: unix:!android: target.path = /opt/$${TARGET}/bin
!isEmpty(target.path): INSTALLS += target`

I think i am missing to linker setup since the error i am getting is:
[image]

OpenCV and QT are both working fine, and are being setup correctly in the qMake file.

Lasy92, 2018-10-05:
To make things easier and more relatable i have now switch over to using cmake instead, my file is looking like this:
```
project(camera_software)
cmake_minimum_required(VERSION 3.1)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_INCLUDE_CURRENT_DIR ON)
set(CMAKE_AUTOMOC ON)

include_directories( ${OpenCV_INCLUDE_DIRS} )

find_package(Qt5Widgets REQUIRED)
find_package(Qt5Core REQUIRED)
find_package(Qt5Gui REQUIRED)
find_package(OpenCV REQUIRED)

set(SOURCES mainwindow.cpp videostreamthread.cpp main.cpp)
set(HEADERS Libraries.h 
		mainwindow.h 
		videostreamthread.h 
		librealsense2/rsutil.h 
    	librealsense2/rs.hpp 
    	librealsense2/rs.h 
    	librealsense2/rs_advanced_mode.hpp 
    	librealsense2/rs_advanced_mode.h 
    	librealsense2/hpp/rs_types.hpp 
    	librealsense2/hpp/rs_sensor.hpp 
    	librealsense2/hpp/rs_record_playback.hpp 
    	librealsense2/hpp/rs_processing.hpp 
    	librealsense2/hpp/rs_pipeline.hpp 
    	librealsense2/hpp/rs_internal.hpp 
    	librealsense2/hpp/rs_frame.hpp 
    	librealsense2/hpp/rs_device.hpp 
    	librealsense2/hpp/rs_context.hpp 
    	librealsense2/h/rs_types.h 
    	librealsense2/h/rs_sensor.h 
    	librealsense2/h/rs_record_playback.h 
    	librealsense2/h/rs_processing.h 
    	librealsense2/h/rs_pipeline.h 
    	librealsense2/h/rs_option.h
    	librealsense2/h/rs_internal.h 
    	librealsense2/h/rs_frame.h 
    	librealsense2/h/rs_device.h 
    	librealsense2/h/rs_context.h 
    	librealsense2/h/rs_advanced_mode_command.h)
    	
set(UIS mainwindow.ui)

set (MACOSX_BUNDLE_INFO_PLIST "${CMAKE_SOURCE_DIR}/my_Info.plist.in") 

qt5_wrap_ui(wrapped_UIS ${UIS})

add_executable(camera_software ${SOURCES} ${HEADERS} ${wrapped_UIS})
target_link_libraries(camera_software Qt5::Core Qt5::Gui Qt5::Widgets  ${OpenCV_LIBS}  )
```

I am still encountering the same linker command failed error. I have looked in to the current used cmakelist.txt files, but with no luck of spotting what i need to add. What target_link_libraries should i add? Does the library realsense2 library need to be installed in some way? 

Thank, Lasse.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@Lasy92 Any other questions for this ticket? Please update. Thanks!

closed 2018-11-27 by RealSense-Customer-Engineering

honghande, 2019-01-09:
@Lasy92 ，You can use that header file in your own project. If you install sdk, you can add attribute configuration to your project. I can tell you the details if you need.

fullprocess, 2019-08-20:
I am using CMake as well (on macOS). What is the library one should link (with link_target_libraries) after including the headers in librealsense/include? Is it librealsense/build/src/gl/librealsense2-gl.dylib?